ropg/ezTime #2: Timezones not working

Spreeuw, 2018-09-21:
A few days ago this was still working for me, but it stopped working for some reason, even with your simple example. when enabling debugging (INFO), I got the following error:
```
Timezone lookup for: Pacific/Auckland
ERROR: Data not found
```
Checking the source, it appears that this library uses https://timezoneapi.io to get the data, which states on their website:

> Announcement: As of October 4th, 2018 TimezoneAPI will no longer support free plans. Read more…

That's still 2 weeks from now, so to get more debugging I set debug to DEBUG, and it looks like they have a 301 redirect from the http address to https:
```HTML
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>301 Moved Permanently</title>
</head><body>
<h1>Moved Permanently</h1>
<p>The document has moved <a href="https://timezoneapi.io/api/timezone/?Pacific%2FAuckland">here</a>.</p>
</body></html>
```

so it looks like going forward another method for getting timezones should be used...

ropg, 2018-09-21:
Meh, well that sucks... Looks like they're forcing https. Which we can do but adds to the code size, esp. on AVRs it's already too tight...

I will investigate soon, but after the weekend. Moving to a different API is possible, could even set up my own I guess.

ropg, 2018-09-25:
Alright: 0.7.4 takes out the dependency on timezoneapi.io and uses the brand new timezoned.rop.nl UDP service that I wrote. Hosted on my own server with 1 Gbps. Has the added benefit of making the AVR code smaller as we now only use UDP and no TCP. Country-code lookups and GeoIP work for countries that do not span multiple timezones. Please test...

closed 2018-09-25 by ropg