ropg/ezTime #3: Possible to use standard NTP servers in your library

mikekgr, 2018-09-27:
Dear Sir,
first of all congratulation for your fine work, by making this library.
My question is: Do you support standard NTP servers in your library? For example, I live in Greece, can I use NTP server like:  gr.pool.ntp.org  ?
Please explain it in details because I am not programmer.

Thanks and Best Regards,
Mike Kranidis

ropg, 2018-09-27:
You'll want to go to the directory of the library. It is called ezTime and is in the `library` subdirectory of wherever your Arduino sketches live. Then go to the subdir `src` of that. In the file `ezTime.h` you will find a line:

```
#define NTP_SERVER				"pool.ntp.org"
```

Change this to whatever server you want and you're in business.

closed 2018-09-27 by ropg

mikekgr, 2018-09-27:
Dear @ropg ,
thanks a lot for your fast support. I had the impression that your library is only work with your own NTP server, now it is clear!!!

Best Regards,
Mike Kranidis