ropg/ezTime #10: Timezone not applying to makeTime

chinswain, 2018-10-26:
Is this expected behavior? I'm sure I'm doing something silly but If I set up a time_t element it prints without timezone adjustment.

The default timezone works, deconstructing t shows the hour as 21 and the epoch time is correct.

```
#include <ezTime.h>
#include <WiFi.h>
Timezone myTZ;

void setup() {
  Serial.begin(115200);
  WiFi.begin("", "");
  waitForSync();

  myTZ.setLocation(F("gb"));
  myTZ.setDefault();
  Serial.print("TimeZone: "); Serial.println(dateTime());
  Serial.print("epoch myTZ : "); Serial.println(myTZ.now());
  
  time_t t = 0;
  t = makeTime(21, 45, 00, myTZ.day() , myTZ.month(), myTZ.year());
  Serial.print("Test: "); Serial.println(myTZ.dateTime(t));

  Serial.print("epoch Test : "); Serial.println(t);

  tmElements_t tmTest;
  breakTime(t, tmTest);
  Serial.print("Hour: "); Serial.println(tmTest.Hour);
}

void loop() {

}
```
```
TimeZone: Friday, 26-Oct-2018 21:45:27 BST
epoch myTZ : 1540590327

Test: Friday, 26-Oct-2018 20:45:00 BST
epoch Test : 1540590300

Hour: 21

```

ropg, 2018-10-27:
`makeTime` is timezone-agnostic on purpose: it just converts between two ways of writing time. So yes, it would represent in UTC. Not sure what the bigger-picture things is you are trying to accomplish, but you can use `tzTime` to convert epoch seconds in one timezone to another.

ropg, 2019-01-22:
I guess when I wrote this I wanted to make it a replacement for Arduino Time library that didn't have timezones, and I didn't want to break code that uses makeTime() and breakTime() by producing unexpected answers to code that isn't timezone-aware.

But I see your point. I've reopened the issue, and as soon as I get some time to sit down with ezTime (early March?), I'll probably implement it the way you suggest. Might come back to you to bounce ideas back and forth at that time, if I may.