ropg/ezTime #9: Daily Alarm?

chinswain, 2018-10-26:
Would it be possible to set a daily repeating alarm without specifying the day, month and year?

I'm using Paul's time and timeAlarm libraries plus some hacky time zone adjustments. (https://github.com/PaulStoffregen/TimeAlarms)

```
 Alarm.alarmRepeat(8,30,0, MorningAlarm);  // 8:30am every day
 Alarm.alarmRepeat(17,45,0,EveningAlarm);  // 5:45pm every day 

// functions to be called when an alarm triggers:
void MorningAlarm(){
  Serial.println("Alarm: - turn lights off");    
}

void EveningAlarm(){
  Serial.println("Alarm: - turn lights on");           
}

```

I'm currently trying with this - which triggers continuously when the time arrives. How should I be setting up a daily repeating alarm?

```
#include <ezTime.h>
#include <WiFi.h>
Timezone myTZ;
void setup() {

  Serial.begin(115200);

  WiFi.begin("", "");

  waitForSync();
  Serial.println();
  Serial.println("UTC:             " + UTC.dateTime());

  myTZ.setLocation(F("gb"));
  Serial.print(F("United Kingdom:         "));
  Serial.println(myTZ.dateTime());
  myTZ.setDefault();

  // Set the event to trigger for the first time
  setEvent( mistOneStart, pumpOn() );
  setEvent( mistOneStop, pumpOff() );
}

void loop() {

  events();

}

void mistOneStart() {
  Serial.print(F("Start! "));
  Serial.println(myTZ.dateTime());
  setEvent( mistOneStart, pumpOn() );
}

time_t pumpOn() {
  int8_t d = myTZ.day() ;
  int8_t m = myTZ.month();
  int16_t y = myTZ.year();

  time_t t = 0;
  t = makeTime(1, 04, 0, d, m, y);
  return t;

}

void mistOneStop() {
  Serial.print(F("Stop! "));
  Serial.println(myTZ.dateTime());
  setEvent( mistOneStop, pumpOff() );
}

time_t pumpOff() {
  int8_t d = myTZ.day() ;
  int8_t m = myTZ.month();
  int16_t y = myTZ.year();

  time_t t = 0;
  t = makeTime(1, 04, 30, d, m, y);
  return t;

}

```

ropg, 2018-10-26:
How about:

```
#include <ezTime.h>
Timezone myTZ;

void setup() {
  waitForSync();
  myTZ.setLocation(F("gb"));
  myTZ.setDefault();
  time_t one_oh_four_am = makeTime(1,4,0, day(), month(), year());
  if (now() >= one_oh_four_am) one_oh_four_am += 24*3600;
  setEvent(mistOneStart, one_oh_four_am);
  setEvent(mistOneStop, one_oh_four_am + 30);
  ...
}

void mistOneStart() {
  setEvent(mistOneStart, now() + 24 * 3600);
  ...
}

void mistOneStop() {
  setEvent(mistOneStop, now() + 24 * 3600);
  ...
}
```

Each function, when called, sets itself to run again 24 hrs later and all should be good. I didn't test, let me know if this works for you...

ropg, 2018-10-26:
I could maybe at some point build a more generic periodic alarm trigger thing. But knowing me, I would then also want to be able to set an alarm for "Monthly, every second Wednesday at 10:00" and such. 

But as for your thing: I'm not sure I understand what the exact remaining problem is: You now have a way of setting an event for a given time and a way for repeating the alarm 24 hrs later. If you want to change the time, simply `deleteEvent` the event and start over. Everything else is just the plumbing in between. Right?

chinswain, 2018-10-27:
It would certainly be appreciated - I'm sure quite a few people will be using this for alarms\triggers.
Something else I was thinking about, is it possible to prevent an event running if it's been missed? I see from your notes past missed events are still triggered?

Would you mind creating an example showing how to trigger an event at the same time each day (Taking the time zone into account) that will not trip up if the trigger time is changed on the fly? With the above, if I remove and recreate the trigger an hour or so before it's scheduled it won't fire on that day.

I have a water pump that needs to come on for xx seconds (user configurable) each day at the same time. I currently perform a DST check and adjust the hour as required.

ropg, 2018-10-29:
Allright: I'll add recurring events sometime soon. I'll think how to make it clean, simple and maximally flexible.

As for an example: I'm not sure I understand: if I check if a time has already happened today and set it for tomorrow instead if it has (like I did above with `if (now() >= one_oh_four_am) one_oh_four_am += 24*3600;`) that covers it, or not?

khawajamechatronics, 2020-02-19:
Hi,

I came here after having issues with TimeAlarm Library. I am trying to understand how library works. Is there examples available for daily alarms?

Thanks

khawajamechatronics, 2020-03-27:
@ropg 
Can I use setEvent like this
setEvent(patternProcess(patternCount), one_oh_four_am);

I want to setup multiple daily alarms. I want them to process some variables when initiate the alarm. Ideally it can be variable passed like patterncount above. 

Also will it setup different alarm based on value of patterncount? 

When I setup like this it seems there is compilation error. 

Is something possible like this ? 

Thanks for your help.

ropg, 2020-03-28:
`setEvent` only takes a function and a time as arguments. The function is essentially just the address of that function in memory. Without raising `MAX_EVENTS` in `ezTime.h`, you have 8 events. You could make 8 event functions, I guess, and then store any values you like to have available for when they trigger separately. I guess I could make it such that it could pass its 8-bit alarm handle back out to the function, that way you know which alarm is being trigered.

I'll leave this issue open, so I'll remember when I get to a bigger update.

robeastwood, 2021-06-19:
Any update on adding functionality for repeating daily events, like for alarms?
I appreciate there are ways of doing, but they seem a bit clunky and manual in comparison to the nice clean functions of the rest of the library.

robeastwood, 2021-06-23:
For anyone else looking for a solution, this is basically what I ended up doing.
It wakes up twice per day, once at the actual alarm time, and again at midnight to determine the right alarm time for that day.
```
void loop() {
  // run events to handle alarms
  events();
}
void handleAlarm(){

  // determine today's alarm time
  time_t alarm = makeTime(alarm_hour,alarm_min,alarm_sec, day(), month(), year());

  // is it time to do something?
  if(now()>=alarm) {

    // do the alarm thing!

    // then wake up again tomorrow
    alarm = makeTime(0,0,0, day(), month(), year())+86400;
  }

  // set next time to wakeup
  wakeUp = setEvent(handleAlarm, alarm);

}
```