ropg/ezTime #15: DST ended early in PT

acortelyou, 2018-11-04:
I noticed that DST ended early in my time zone on version 0.7.7.

Possibly related to #11?

Using `setLocation("America/Los_Angeles")`:

|Date|Time|ezTime|
|----:|-----:|-------:|
|NOV 3|11:00PM|10:00PM
|NOV 4|12:00AM|11:00PM|
|NOV 4|1:00AM|12:00AM|
|NOV 4|1:00AM|1:00AM|
|NOV 4|2:00AM|2:00AM|

ropg, 2018-11-04:
Drats...  I can see it started showing the right time at 1 AM, but I can't from your table see when it started showing the wrong time. Will figure this out tomorrow...

acortelyou, 2018-11-04:
To clarify: the time indicated by ezTime was linear for the duration and only became correct as DST ended and the actual time changed to match.

Unfortunately I do not know when ezTime became incorrect initially nor if this behavior was introduced with 0.7.7.

ropg, 2019-03-25:
I believe this is fixed by the fix for #38, my makeOrdinalTime function was off when the last of a given weekday in a given month was requested. Please reopen if problem reoccurs.

closed 2019-03-25 by ropg