ropg/ezTime #18: Please document privacy policy

zobelhelas, 2018-12-08:
You are offering a service on timezoned.rob.nl so users of your library can find out DST and TZ information.

Please document your privacy policy for information collected on this service as per EU GDPR.

ropg, 2018-12-09:
Eh...

It's not really in a finished state, but yes, there is data. There is a logfile, that I look at irregularly to see if everything is running smoothly and to rejoice in the fact that there are real-world users. Every so often (whenever I have time to look at it) I might throw out the old logfile and start a new one. Data has not been used for anything but that, and I don't envision ever using it for anything else.

I am not a lawyer. Where would you suggest I put what document to say this?

chinswain, 2018-12-09:
I don't think you are handling any pii? (Personally identifiable information) other than IP addresses, I guess a short description in the readme on what data is collected + a fixed time to purge the log (if it contains pii) will satisfy those that are bothered by this.

[https://eugdprcompliant.com/personal-data/](https://eugdprcompliant.com/personal-data/)

"A much discussed topic is the IP address. The GDPR states that IP addresses should be considered personal data as it enters the scope of ‘online identifiers’. Of course, in the case of a dynamic IP address – which is changed every time a person connects to a network – there has been some legitimate debate going on as to whether it can truly lead to the identification of a person or not. The conclusion is that the GDPR does consider it as such."

ropg, 2018-12-09:
I deleted the log and started without, can always turn it on before debugging. The README now has the following new chapter:

&nbsp;

> ### timezoned.rop.nl
> 
> `timezoned.rop.nl` is ezTime's own timezone service that it connects to. It is a simple UDP service that gets a packet on UDP port 2342 with the request, and responds with a packet that holds the POSIX information for that timezone (after `OK `) or the error (after `ERR `). It will only respond to the same IP-number once every three seconds to prevent being used in dDoS attacks.
> 
> The service has the potential of seeing which IP-numbers use ezTime and what timezone data they request. Any GeoIP lookups are done against a local database, no third parties are involved. The service does not keep logfiles unless something is wrong and needs debugging. In such a case any logfiles will be deleted after work is done, but within 48 hours at the latest.
> 
> Data has never been used for any other purposes than debugging, not will it ever be.

closed 2018-12-09 by ropg

zobelhelas, 2018-12-09:
Thanks for doing so, and responding to this issue while i was sleeping. Your fast response is very much appreciated.