ropg/ezTime #22: Can I use this library also without Internet connection?

x0tester0x, 2019-01-11:
So for example can I use a local NTP server?

ropg, 2019-01-11:
Yes you can... Simply point it to the local NTP server and do not use `.setLocation`. (That would try to connect to my timezone info server.)

That means you need to feed it timezone information, either as a static offset to UTC or as a 'Posix' string that has information about when to switch from Standard to Daylight Savings Time. Use the `.setPosix` method. ([User Manual](https://github.com/ropg/ezTime/#timezones-1))

You can also use ezTime without any network at all, for instance if you have a time source that is not NTP, such as a hardware RTC device or a GPS receiver. In that case use `setInterval()` to turn off NTP updating and use `.setTime` ([User Manual](https://github.com/ropg/ezTime/#settime)) to set/update the time yourself. Or, if you are on a memory constrained platform such as AVR Arduinos, do not even compile networking support into ezTime altogether by commenting out `#define EZTIME_NETWORK_ENABLE` in `ezTime.h` in the library's `src` directory. ([User Manual](https://github.com/ropg/ezTime/#smaller-footprint-avr-arduinos))

closed 2019-01-11 by ropg

x0tester0x, 2019-03-17:
Can you add an example for this please?