ropg/ezTime #23: timeStatus timeNeedsSync never seted

dicson, 2019-01-13:
Thanks for you work!
I think timeNeedsSync status never can be set. I did not find in the code timeNeedsSync and if I disable connection I always see status '2' (timeSet).
Maybe I'm wrong. Sorry for my bad english

ropg, 2019-02-06:
Sorry for the late reply. Will leave this open and investigate soon as I get some time to work on it.

mike-s123, 2019-02-08:
I think you want something like this at the end of updateNTP():

```
      if (queryNTP(_ntp_server, t, measured_at)) {
...
      } else {
        if ( nowUTC() > _last_sync_time + _ntp_interval + NTP_STALE_AFTER ) {
          _time_status = timeNeedsSync;
        }  
        UTC.setEvent(updateNTP, nowUTC(false) + NTP_RETRY);
      }

```

ropg, 2019-02-10:
Fixed by commit 16539554d790465eececab97040cf6262331b8b8 from #29 (thanks!)

closed 2019-02-10 by ropg

dicson, 2019-02-10:
Thank you very much!