ropg/ezTime #30: Feature request: expose _last_sync_time

mike-s123, 2019-02-10:
It's great that it's logged in debug output, but it would be useful to be able to know programmatically when the last good network update occurred. Perhaps time_t lastNtpUpdateTime()

ropg, 2019-02-10:
You wanna do this and make it a pull request or do I make it?

mike-s123, 2019-02-10:
As I mentioned, I'm not that much of a coder. I can figure out logic errors and do basic stuff. But properly creating new functions in an object-oriented environment, etc., not so much.

keith0603, 2019-03-13:
I would like to see this as well,  but like mike-s123, don't have the knowledge to do the pull request myself. To add context, I am using the code on an esp8266 as a time-keeper feeding into a arduino that is running additional code. I want to keep the interrupts to the arduino a minimum, so only wanted to update after a NTP sync. If the internal NTP sync event were exposed, it seems an easy way to accomplish this. Great library and lives up to it's name! Thank you.

ropg, 2019-03-26:
Added in 0.7.10

closed 2019-03-26 by ropg