ropg/ezTime #35: Question: ezTime::setTime() and Arduino::configTime() not the same

ksaye, 2019-03-14:
While I love ezTime, I am confused how differently it works on an ESP8266 vs the configTime().

While writing in Arduino on a NodeMCU, I falsely assumed that calling ezTime::setTime() would set the system time that the Azure IoT Client SDK would honor.  I still had to call the configTime() which grabbed from an NTP service.

From a core perspective, what is the difference between Arduino.h::configTime() and setTime()?

ropg, 2019-03-19:
configTime is Espressiv-specific, and not available on all Arduino hardware, whereas ezTime is. Also the time library on ESPs is not as full-featured (although it does support a timezone as well as NTP, so it may well be enough for you.)

closed 2019-03-19 by ropg

khawajamechatronics, 2020-04-04:
@ropg I am having same issue, can anything be done to make it compatible?

ropg, 2020-04-04:
configTime is ESP specific. Use it if you like, it does (sort of) the same thing as ezTime. You may not need both, they are independent.

khawajamechatronics, 2020-04-04:
I have got it working somehow, we only can use UTC time with SDK and I had to minus some seconds to get it work.