ropg/ezTime #40: waitForSync is not returning true

nospam2k, 2019-03-22:
Sometimes, after doing a firmware update. waitForSync won't return true. Here is my code.

Timezone US;
...

void setup()
{
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  Serial.println("Connecting to Wifi");

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  Serial.println("Connected to " + String(ssid) + " on address: " + WiFi.localIP().toString());

  while (!waitForSync(3))
  {
   Serial.println("Retrying timeSync");
  }
  Serial.println("Got timeSync");
}

void loop() {}

ropg, 2019-03-26:
Can you retry this? I set the NTP error-retry back to 20 seconds.

nospam2k, 2019-04-17:
Sorry for delay. Because of the bug I mentioned, I've just setup to reset the ESP8266 until it gets the time as I don't really have time to test it. Waiting 20 seconds isn't really an option anyway as when it does connect it usually does so within 3 seconds so I would just leave the time reset code anyway. Feel free to close this if you wish.

vbbalaji, 2019-06-15:
Im having the same problem. It has reconnected almost 15 times with 20s retry and Im not getting anything back. 
I have tried multiple times and without the loop. Luckily sometime it gets the time quickly!! Surprisingly connecting to the ntp server directly using a code I found in the net works every time. This probably needs to be retested?

`    delay(10);
    // We start by connecting to a WiFi network
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    Serial.println(IPAddress2String(WiFi.localIP()));

    Serial.print("Connecting");
    while (IPAddress2String(WiFi.localIP()) == "0.0.0.0") {
      delay(1000);
      Serial.print(".");
    }
    Serial.println("WiFi connected");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
    Serial.println("Synching Time Server...");
    int attemptCnt = 1;
    while (!waitForSync(20))
    {
      attemptCnt++;
      Serial.println("Reconnecting...");
    }
    Serial.println("UTC: " + UTC.dateTime());
    myTZ.setLocation(F("America/New_York"));
    Serial.print("My Local Time:");
    Serial.println(myTZ.dateTime());`

ropg, 2019-06-15:
I will have a look soon. Can you tell me what Arduino board this is ran on?

vbbalaji, 2019-06-15:
ESP8266 - nodeMCU

ernfel, 2020-03-10:
I experience exactly the same like @vbbalaji. Sometimes it just doesnt get the time back. 
Board: esp32 (heltec wifi kit lora 32)

rvdbreemen, 2020-08-11:
I have run into the fact that time is not synced in the waitforsync()... then after a while it "just syncs". I think I have fixed it now by using updateNTP() before waitforsync, but somehow I was not expecting this. I am using a esp8266 on a custom PCB design.

josinSbazin, 2024-11-20:
Just use other ntp server ( for ex. "ntp21.vniiftri.ru").
Set it like this:
`
    setServer("ntp21.vniiftri.ru");
    waitForSync();
`

pringtest, 2025-05-12:
> By default in `ezTime.h` lib use pool.ntp.org server. `#define NTP_SERVER "pool.ntp.org"` In my experience this servers is unavailable sometimes. Just use other ntp server ( for ex. "ntp21.vniiftri.ru"). Set it like this:
> 
> ```
>     setServer("ntp21.vniiftri.ru");
>     waitForSync();
> ```

thanks it working like a charm!