ropg/ezTime #41: All my clock based on this library started to show one hour more today

BugerDread, 2019-03-24:
I will write more details later as now I have different thing to do, but all my clock based on ESP8266 and this library are showing +1 hour (now is 12:47 here, clock shows 13:47). Im Europe / Prague TZ, manually set (not geoIP), tried to restart and received wrong time again.

BugerDread, 2019-03-24:
maybe wrong info about our TZ because "summer time" starts next week here
http://www.timezoneconverter.com/cgi-bin/zoneinfo?tz=Europe/Prague

Tederean, 2019-03-24:
Same problem here. Using Europe/Berlin shows active DST even though it's not 31-March or beyond.

ropg, 2019-03-25:
I will release a new version of ezTime, hopefully tomorrow.

ropg, 2019-03-25:
I believe this is fixed by the fix for #38, my makeOrdinalTime function was off when the last of a given weekday in a given month was requested. Please reopen if problem reoccurs.

closed 2019-03-25 by ropg