ropg/ezTime #42: eztime does not use BST correctly

malbrook, 2019-03-26:
Today 26th March the time displayed using the ezTime library is one hour ahead for timezone United Kingdom, unfortunately BST does not start in the UK until the last Sunday in March. Last year it was 25th March, this year it is 29th March and next year it will be different again. Is eztTime designed to compensate for this change in the starting (and also ending) dates for BST?

ropg, 2019-03-26:
Have you upgraded to version 0.7.10 of ezTime? If not, please do so and see if effect persists. Please also report if that fixes it.

malbrook, 2019-03-26:
Hi Ron
That was an impressive response time, I was using 0.7.9, upgraded to 0.7.10 and it cleared up the problem, would you be good enough to explain the change?

best regards
Malcolm Brook

closed 2019-03-26 by malbrook

ropg, 2019-03-26:
It's all to do with an issue with `makeOrdinalTime`, the function in ezTime that can give you a time for 'What is the last Sunday in April`. It screwed up in cases where the last weekday of a given month was involved, such as in the current time switch. Read all about it in Issue #38 on this repository if you want to know more.