ropg/ezTime #43: Timezone definition incorrect

agoston, 2019-03-28:
Hi,

Awesome library, however, this week, a bug was revealed in timezone config.
In the README.md, and also in your lookup server, it is stated that timezone for Berlin/western eu is:

```
CET-1CEST,M3.4.0/2,M10.4.0/3
```

However, the actual definition is 'last sunday of the month'. In march 2019, the month had 5 sundays, so the clock on my night table is off for an hour this week.

According to the definition, '5' means 'last of the month', so I've also took the liberty to update the definition in my code to

```
CET-1CEST,M3.5.0/2,M10.5.0/3
```

Can you update README.md and probably your server accordingly?

ropg, 2019-03-28:
If you upgrade to 0.7.10, you'll likely notice the problem is fixed...

ropg, 2019-03-28:
Fixed the documentation now also...

closed 2019-03-28 by ropg

agoston, 2019-03-29:
drat, i did check the commits, but missed the "last week patch" comment, sorry.
Good to know it's fixed, thanks!