ropg/ezTime #46: lastNtpUpdateTime() incorrect.

mike-s123, 2019-04-10:
Shortly after things get started and NTP is updated,

myTz.dateTime(RFC850) produces "Wednesday, 10-Apr-2019 10:30:37 EDT"

but

dateTime(lastNtpUpdateTime(),RFC850) produces "Wednesday, 10-Apr-2019 18:29:37 EDT"

Seems to be a sign issue. EDT is UTC-4, and the reported time is off by 8.

mitag, 2019-04-10:
Could this have something to do with this?
"If you execute multiple calls to setLocation, make sure they are more than 3 seconds apart, because the server will not answer if calls from the same IP come within 3 seconds of one another (see below)."
During a test I recalled the time every second and was always shown CEST and UTC alternately, which makes a difference of 2 hours.

mike-s123, 2019-04-10:
Nope. I use setPosix (so there's no dependency on external servers), and it's only called once, during initialization. And, the data above was after a single poll of the ntp server.

ropg, 2019-04-10:
Strange... I will have a look at this soon.

mike-s123, 2019-12-22:
Bug still present in 0.8.3.

    myTZ.dateTime(RFC850) returns Sunday, 22-Dec-2019 16:01:05 EST 
    myTZ.dateTime(lastNtpUpdateTime(),RFC850) returns Sunday, 22-Dec-2019 20:59:34 EST

The former is correct, the latter is off by +5 hours (these were shown shortly after a boot and fresh NTP sync).

ropg, 2019-12-23:
lastNtUpdateTime is in UTC, which I guess you are 5 hours away from.

Try `myTZ.dateTime(lastNtpUpdateTime(), UTC_TIME, RFC850)`

mike-s123, 2019-12-23:
OK, that worked. Thanks.

Please update the docs - https://github.com/ropg/ezTime#datetime - which show dateTime only taking 2 arguments. ("String dateTime(TIME, String format = DEFAULT_TIMEFORMAT);")

ropg, 2019-12-23:
It's there. Check out the [section 'specifying time'](https://github.com/ropg/ezTime/#specifying-time) in the manual...

closed 2019-12-23 by ropg