ropg/ezTime #47: SetLocation for "ie", "uk" and "pt" does not work

witnessmenow, 2019-04-14:
Using the timezones examples as base, none of the above return the correct time. They display as UTC, when they should be now UTC + 1

When you enable info serial prints, you see that none of them return a response for the timezone lookup. Setting the country code to "de" works as expected (UTC +2) and returns a response to the timezone lookup.

Setting the timezone in the olson format works as expected ("Europe/Dublin" returns UTC +1)

emeyedeejay, 2019-04-14:
I believe they only work for countries with a single time zone. I had the same with ZA but the Olson strings work too. Does GEO IP lookup work for you? I would think not for the same reason.

ropg, 2019-04-14:
For some reasons the country lookups do not work anymore, except for a few that are programmed exceptions, such as "de" (Don't ask). I'll keep this issue open and have a look at it later.

Either way, the preferred way to do a timezone setting is by the olson string, so for the UK that would be "Europe/London" (or any unique substring thereof, such as "London") See [here](https://en.wikipedia.org/wiki/List_of_tz_database_time_zones) for a list of these timezone identifier strings.

emeyedeejay, 2019-04-15:
Is this the same reason GeoIP doesn't work as well? That is a fantastic feature!

Thanks for this library - it's amazing how much it does!

ropg, 2019-10-29:
I fixed this issue. Sorry it took a long time...

closed 2019-10-29 by ropg