ropg/ezTime #50: Seem good, but do not work !?

azimut2000, 2019-05-10:
Hello,

I have this code : 

`  Timezone myTZ;
  myTZ.setLocation(F("CET"));
  Serial.print(F("CET:\t\t\t"));
  Serial.println(myTZ.dateTime());
  myTZ.setLocation(F("CEST"));
  Serial.print(F("CEST:\t\t\t"));
  Serial.println(myTZ.dateTime());
  myTZ.setLocation(F("UTC"));
  Serial.print(F("UTC:\t\t\t"));
  Serial.println(myTZ.dateTime());
  myTZ.setLocation(F("Europe/London"));
  Serial.print(F("London:\t\t\t"));
  Serial.println(myTZ.dateTime());
  myTZ.setLocation(F("Europe/Paris"));
  Serial.print(F("France:\t\t\t"));
  Serial.println(myTZ.dateTime());
  myTZ.setPosix("CET+2:00");
  Serial.print(F("France apres ofset:\t"));
  Serial.println(myTZ.dateTime());
  Serial.println("Posix zone : " + myTZ.getPosix());
  Serial.println("Nom de la zone : " + myTZ.getTimezoneName());

`
It's very strange, the result is :

> CET:			Friday, 10-May-2019 23:29:09 UTC
CEST:			Friday, 10-May-2019 23:29:11 UTC
UTC:			Friday, 10-May-2019 23:29:13 UTC
London:			Friday, 10-May-2019 23:29:15 UTC
France:			Friday, 10-May-2019 23:29:17 UTC
France apres ofset:	Friday, 10-May-2019 21:29:17 CET
Posix zone : CET+2:00
Nom de la zone : CET

I do not understand why it's the same hour whatever the location i take !?!?

More fun : actually, there is 2 hours offset as we are in summer (only 1 hour in winter) : so why does it send me 21:29 whereas it's 01:29 am ???
Why is it the same hour in Paris and London whereas it's not the same time zone ???

Does someone have an idea ? or what do i make wrong ? 
Thank's again,
Thierry

ropg, 2019-05-11:
You're probably running into the rate limiting on the `timezoned.rop.nl` timezone server, which I had to put in to make sure the UDP-based server cannot be abused to amplify dDoS attacks.

From the manual:

> If you execute multiple calls to setLocation, make sure they are more than 3 seconds apart, because the server will not answer if calls from the same IP come within 3 seconds of one another (see below).
>
> ### timezoned.rop.nl
>
> timezoned.rop.nl is ezTime's own timezone service that it connects to. It is a simple UDP service that gets a packet on UDP port 2342 with the request, and responds with a packet that holds the POSIX information for that timezone (after OK ) or the error (after ERR ). It will only respond to the same IP-number once every three seconds to prevent being used in DDoS attacks.

This still does not explain why the first lookup didn't work, unless your internet wasn't fully up by then. Please put in some delays and let us know if it works then.

azimut2000, 2019-05-11:
Thank's for answer (and sorry for my bad English)

I've put a function at the beginning in order to wait good connection with : 

`while` (WiFi.status() != WL_CONNECTED) `{}`

I've also tried with delay between 30s and up to 3 minutes and nothing is better.

I've tried other things :  it's not normal that that CET time give me 2 hours in late whereas i live in France. I do not understand as well why i must put "-2:00" with CET time in order to have the good hour : France is CET+1 and as we are in summer time, we must put again 1 hour more in order to have good time ? 
I tried with myTZ.setLocation(F("Europe/Paris")) and myTZ.setLocation(F("Fr")), but it's not good as well.

ropg, 2019-05-11:
Je parle Français aussi, mais pour les autres c'est peut-etre mieux de parler Anglais ici... :)

As to the way to write Posix strings: it's not my idea to write the offset in Posix timezone notation the other way around from most other timezone notations. From the ezTime [manual](https://github.com/ropg/ezTime/#timezones-1):

> Hours offset from UTC, meaning subtract one hour from this time to get to UTC. (Note offset is often written elsewhere the other way around (so +1 in this case), just to confuse things.) Could also specify minutes, like -05:30 for India.

As to your other problem: I just tested my timezone server and it is providing correct answers:

```
$ nc -u timezoned.rop.nl 2342
ParisOK Europe/Paris CET-1CEST,M3.5.0,M10.5.0/3
```

ezTime works for me. I am currently assuming that you have some problem reaching my timezone server. What does `ping timezoned.rop.nl` output for you?

azimut2000, 2019-05-12:
Hehe... Oui, nous allons continuer en anglais ;-)

Here the command result : 

`thierry@pine64:~$ ping timezoned.rop.nl

PING timezoned.rop.nl (84.200.21.154) 56(84) bytes of data.

64 bytes from rop.nl (84.200.21.154): icmp_seq=1 ttl=54 time=19.1 ms

64 bytes from rop.nl (84.200.21.154): icmp_seq=2 ttl=54 time=18.5 ms

^C

--- timezoned.rop.nl ping statistics ---

2 packets transmitted, 2 received, 0% packet loss, time 3004ms

rtt min/avg/max/mdev = 18.506/19.021/19.627/0.426 ms

root@pine64:~# nc -u -4 -D timezoned.rop.nl 2342

^C
`

I finally stop the request because nothing come...
I've thought that it could be a proxy or router in my LAN that could make a problem, but for the test above, i've open my computer on the DMZ and it do the same.

Are you sure of your server ? when i test it on port 2342 with UDP frame on the service https://portscanner.standingtech.com, it tell me : The port is close. [timezoned.rop.nl:2342]

Strange...

ropg, 2019-05-12:
As soon as you have entered the nc command type "Paris" (without the quotes) and then Ctrl-D

azimut2000, 2019-05-12:
(sorry :-( )
root@pine64:~# nc -u timezoned.rop.nl 2342
Paris
ERROR Timezone Not Found

ropg, 2019-05-12:
No return after Paris, just the Ctrl-D ? (You are talking to the server though, because it managed to say "Timezone Not Found")

azimut2000, 2019-05-12:
root@pine64:~# nc -u timezoned.rop.nl 2342
ParisOK Europe/Paris CET-1CEST,M3.5.0,M10.5.0/3

ropg, 2019-05-12:
OK, so that works. You will need to tell me more about the whole setup there. What Arduino are you using? How is connected? Can you paste the whole sketch?

azimut2000, 2019-05-12:
It's an ESP8266, connected on my WiFi; here is the full code : 

`#include <ezTime.h>
#include <ESP8266WiFi.h>

void setup() {

	Serial.begin(115200);
	
	WiFi.begin("rezoprive", "thierrybenetboissy");
	while (!Serial) { ; }		// wait for Serial port to connect. Needed for native USB port only

  // On attend d'être connecté
  unsigned long MyTime = millis();
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (millis() >= ((20*1000) + MyTime) ) {  // si au bout de 20 secondes la connexion n'est pas OK...
      Serial.println("");
      Serial.println("On a dépassé le temps imparti pour la connexion : reboot !");
      ESP.restart();                          // ... on reboot !
    }
  }
  Serial.println("");

  waitForSync();

  Timezone myTZ;

  myTZ.setLocation(F("CET"));
  Serial.print(F("CET:\t\t\t"));
  Serial.println(myTZ.dateTime());

  delay(30000);

  myTZ.setLocation(F("CEST"));
  Serial.print(F("CEST:\t\t\t"));
  Serial.println(myTZ.dateTime());

  delay(30000);

  myTZ.setLocation(F("UTC"));
  Serial.print(F("UTC:\t\t\t"));
  Serial.println(myTZ.dateTime());
  Serial.println("UTC.dateTime(): \t" + UTC.dateTime());

  delay(30000);

  myTZ.setLocation(F("Europe/London"));
  Serial.print(F("London:\t\t\t"));
  Serial.println(myTZ.dateTime());

  delay(30000);

  myTZ.setLocation(F("Europe/Paris"));
  Serial.print(F("France:\t\t\t"));
  Serial.println(myTZ.dateTime());

  delay(30000);

  myTZ.setPosix("CET-2:00");
  Serial.print(F("France apres ofset:\t"));
  Serial.println(myTZ.dateTime());

  Serial.println("Posix zone : " + myTZ.getPosix());
  Serial.println("Nom de la zone : " + myTZ.getTimezoneName());
  Serial.println("Offset : " + String(myTZ.getOffset()));

  setInterval(20);
  
}

void loop() {
	events();
}`

Very simple... ;-)

ropg, 2019-05-12:
Will try to play tomorrow if I have time. For now: try removing the F("...") structures (just a hunch, and not needed on ESP, compiler smart enough). Also, you cannot use `setLocation` with time zone abbreviations, just "Continent/City" style (but any match works, as you saw with just "Paris"). Also, delay of 5 secs should always be enough. And I would be curious what this all outputs if you put a `setDebug(INFO);` somewhere in `setup()`. Off to bed now...

azimut2000, 2019-05-12:
I've just tried again (the same code as above) : it work now !!!
I've just remove the F("..") structure... and it work... but with or without it, it work as well !!!

Here is what i get : 

`.......

ezTime debug level set to INFO
Waiting for time sync
Querying pool.ntp.org ... success (round trip 37 ms)
Received time: Sunday, 12-May-19 22:23:26.848 UTC
Time is in sync
Timezone lookup for: CET ... (round-trip 44 ms)  CET:			Sunday, 12-May-2019 22:23:26 UTC
Timezone lookup for: CEST ... (round-trip 25 ms)  CEST:			Sunday, 12-May-2019 22:23:31 UTC
Timezone lookup for: UTC ... (round-trip 23 ms)  success.
  Olson: Etc/UTC
  Posix: UTC0
UTC:			Sunday, 12-May-2019 22:23:36 UTC
UTC.dateTime(): 	Sunday, 12-May-2019 22:23:36 UTC
Timezone lookup for: Europe/London ... (round-trip 22 ms)  success.
  Olson: Europe/London
  Posix: GMT0BST,M3.5.0/1,M10.5.0
London:			Sunday, 12-May-2019 23:23:41 BST
Timezone lookup for: Europe/Paris ... (round-trip 22 ms)  success.
  Olson: Europe/Paris
  Posix: CET-1CEST,M3.5.0,M10.5.0/3
France:			**Monday, 13-May-2019 00:23:47 CEST**
France apres ofset:	**Monday, 13-May-2019 00:23:52 CET**
Posix zone : CET-2:00
Nom de la zone : CET
Offset : -120
Querying pool.ntp.org ... success (round trip 13 ms)
Received time: Sunday, 12-May-19 22:24:11.002 UTC (internal clock was 11 ms slow)
Querying pool.ntp.org ... success (round trip 11 ms)
Received time: Sunday, 12-May-19 22:24:31.012 UTC (internal clock was 1 ms fast)
Querying pool.ntp.org ... success (round trip 11 ms)
Received time: Sunday, 12-May-19 22:24:51.011 UTC (internal clock was spot on)
`

I do not understand why it work now... I do not change anything, only put an "setDebug(INFO);" ! 
Thank's for your help and hope it will be always good tomorrow ;-)
Going to sleep as well... Good night !

ropg, 2019-05-13:
All is well that ends well... Glad it works, closing the issue.

closed 2019-05-13 by ropg