ropg/ezTime #52: error connecting to pool.ntp.org

callumjackson, 2019-06-11:
hi, trying to get this to work but everthing i have tried leaves me with this.

`ezTime debug level set to INFO
Waiting for WiFi ... Querying pool.ntp.org ... ERROR: No network
connected
Waiting for time sync
Querying pool.ntp.org ... ERROR: Timeout
Querying pool.ntp.org ... ERROR: Timeout
`

ropg, 2019-06-11:
You'll have to tell me a bit more about your setup, and how you are connecting to Wifi. 

On first glance it looks like you can connect to Wifi but not reach pool.ntp.org.

callumjackson, 2019-06-11:
hi, I am using nodeMCU as a basis and was testing this to see if I could get the time to output to a database as a timestamp, I'm connected to a wifi network but can't connect to pool.ntp.org. the time I wish to use is GMT as I live in the UK and have tried connecting to the uk.pool.ntp.org server but I get the same error

pksiazek, 2019-07-13:
I had same issue.
I didn't have DNS server configured for my WiFi connection (I used static IP).
Maybe that's your case?

eskhisov, 2019-11-09:
I am getting the same error, but only  about half the time:

```
ezTime debug level set to DEBUG
Waiting for time sync
Querying pool.ntp.org ... ERROR: Timeout
Set event (#1) to trigger on: Thursday, 01-Jan-1970 00:00:27 UTC
```
The other times, it connects just fine.

underwoodblog, 2019-11-10:
@callumjackson have you checked if you're connected? Can you ping ntp.pool with your computer?

```
      Serial.println();
      Serial.printf("Connection status: %d\n", WiFi.status());
      Serial.print("Connected to    : ");
      Serial.println(WiFi.SSID());
      Serial.print("IP address      : ");
      Serial.println(WiFi.localIP());
      Serial.print("MAC Address     : ");
      Serial.println(WiFi.macAddress());
      Serial.print("Signal strength : ");
      Serial.println(WiFi.RSSI());
```

eskhisov, 2019-11-10:
It is connected in both cases:

```
Connection status: 3
Connected to    : CenturyLink6487
IP address      : 192.168.0.24
MAC Address     : 5C:CF:7F:8B:24:39
Signal strength : -74

ezTime debug level set to DEBUG
Waiting for time sync
Querying pool.ntp.org ... ERROR: Timeout
Set event (#1) to trigger on: Thursday, 01-Jan-1970 00:00:31 UTC
```

```
Connection status: 3
Connected to    : CenturyLink6487
IP address      : 192.168.0.24
MAC Address     : 5C:CF:7F:8B:24:39
Signal strength : -75

ezTime debug level set to DEBUG
Waiting for time sync
Querying pool.ntp.org ... Received data:
0: 24, 2, 9, E9, 
4: 0, 0, 0, 5B, 
8: 0, 0, B, AE, 
12: D8, DA, FE, CA, 
16: E1, 72, EF, 4E, 
20: 67, 8D, 60, D7, 
24: 0, 0, 0, 0, 
28: 0, 0, 0, 0, 
32: E1, 72, F6, EC, 
36: 74, A3, AC, 73, 
40: E1, 72, F6, EC, 
44: 74, B1, C3, 83, 
success (round trip 64 ms)
Received time: Sunday, 10-Nov-19 20:51:56.489 UTC
Set event (#1) to trigger on: Sunday, 10-Nov-2019 21:21:57 UTC
Time is in sync
```

Drachenkaetzchen, 2019-11-17:
Same issue here - even with the direct IP address or any other ntp server it's the same problem. Sometimes(tm) it works, but often it's not.

ropg, 2019-11-17:
Maybe the timeout is too tight? Any of you on particularly slow links? (cellphone tether? Satellite?) Could you edit ezTime.h and on line 145 change the `#define NTP_TIMEOUT` from 1500 to 3000 ?

Drachenkaetzchen, 2019-11-17:
> Maybe the timeout is too tight? Any of you on particularly slow links? (cellphone tether? Satellite?) Could you edit ezTime.h and on line 145 change the `#define NTP_TIMEOUT` from 1500 to 3000 ?

I did that, no change. However, I observed that when I use a locally setup NTP server it works fine. So that workaround does the trick for me.

ropg, 2019-11-18:
Hmmmmm... 🤔

My code has no way of knowing what is local or not, it just sends and receives UDP packets. If it's not the round-trip time I would almost suspect something is wrong with the DNS setup, or something else is causing UDP (or all IP) to be flaky.

ropg, 2019-11-18:
It might well be related to problems some people are experiencing with ESP-32s more broadly. (Such as [this one](https://github.com/espressif/arduino-esp32/issues/2778), but there's many other issues.)

Can you compile with debugging info on? The underlying ESP software might then actually say things on the serial port like `[E][WiFiGeneric.cpp:658] hostByName(): DNS Failed` if it's a DNS problem. Or something else useful, who knows...

eskhisov, 2019-11-23:
There does not seems to be a DNS issue.  Here is Debug info:
```
Waiting for time sync
Querying pool.ntp.org ... [hostByName] request IP for: pool.ntp.org
[hostByName] Host: pool.ntp.org IP: 72.30.35.89
```

It always resolves the host name to IP address.  There seems to be myriad of IP addresses.  Some of them connect all the time, some always time out. And some connect half the time and time out the rest.

I tried to add a loop:
```
while (!waitForSync(6))
  {
   Serial.println("Retrying timeSync");
  }
```
But it does not really loops.  When it calls waitForSync() second time, it just hangs.

ropg, 2019-11-24:
Hmmm. I'd have to look at the code to see why it would hang if called a second time.

pool.ntp.org indeed returns one of many IP addresses. It could be that some just barely  time out on a slow link and some don't. 

Have you also tried extending the NTP timeout like above:

>Maybe the timeout is too tight? Any of you on particularly slow links? (cellphone tether? Satellite?) Could you edit `ezTime.h` and on line 145 change the `#define NTP_TIMEOUT` from 1500 to 3000 ?

monarchit, 2020-07-24:
For some reason many NTP servers are giving me slow responses quite often. I solved this by setting:

`setServer("time.cloudflare.com");`

Works incredibly well compared to good old pool.ntp.

If you are not close to one of their pool ntp servers or live in an obscure country like me then you might also get some random timeouts. :)

https://blog.cloudflare.com/secure-time/

877dev, 2020-07-31:
> For some reason many NTP servers are giving me slow responses quite often. I solved this by setting:
>
> `setServer("time.cloudflare.com");`
>
> Works incredibly well compared to good old pool.ntp.
>
> If [...]

This worked for me, thank you!

I was getting intermittent connection issues, now it connects immediately.

cgleon809, 2021-01-19:
Hi!

Jonny13xx, 2021-01-21:
Hi.

I'm using ESP8266WiFi and ezTime.

With DHCP everything works fine !

When I use static IP, also with same IP address I've got from DHCP, I get timeout from pool.ntp.org
and other NTP servers with error code  :

**[hostByName] request IP for: time.cloudflare.com
[hostByName] Host: time.cloudflare.com lookup error: -6!**

In both cases network itself works without problems.

Any idea ?

Erik

monarchit, 2021-01-21:
> Hi.
>
> I'm using ESP8266WiFi and ezTime.
>
> With DHCP everything works fine !
>
> When I use static IP, also with same IP address I've got from DHCP, I get timeout from pool.ntp.org
> and other NTP servers [...]

What code are you using to set the static IP address? It seems the DNS is not being set correctly so it can't resolve an IP for that hostname.

monarchit, 2021-01-21:
> > > Hi.
> > > I'm using ESP8266WiFi and ezTime.
> > > With DHCP everything works fine !
> > > When I use static IP, also with same IP address I've got from DHCP, I get timeout from pool.ntp.org
> > > and other NTP servers with error code :
> > > **[hostByName] request IP for: time.cloudflare.com [hostByName] Host: time.cloudflare.com lookup error: -6!**
> > > In both cases network itself works without problems.
> > > Any idea ?
> > > Erik
> > 
> > 
> > What code are you using to set the static IP address? It seems the DNS is not being set correctly so it can't resolve an IP for that hostname.
> 
> Thx, Monarchit.
> 
> That was my idea too, but I don't find right procedure for DNS setting...
> 
> I tested meanwhile "WiFi.hostByName(ntpServerName, timeServer)" but without success
> 
> Erik

Try to follow the instructions on [this link](https://randomnerdtutorials.com/esp8266-nodemcu-static-fixed-ip-address-arduino/) to set a fixed IP address and the rest of the options. The example shows:

```
IPAddress local_IP(192, 168, 1, 184);
IPAddress gateway(192, 168, 1, 1);

IPAddress subnet(255, 255, 0, 0);
IPAddress primaryDNS(8, 8, 8, 8);
IPAddress secondaryDNS(8, 8, 4, 4);
```
Based on this you can either use your router's IP address as a primary DNS or Google's public DNS listed in the example. I believe you are missing the rest of the configuration.

Jonny13xx, 2021-01-21:
> Try to follow the instructions on [this link](https://randomnerdtutorials.com/esp8266-nodemcu-static-fixed-ip-address-arduino/) to set a fixed IP address and the rest of the options. The example [...]

WOW !
I forgot to extend the config call ... 

_**WiFi.config(ip, gateway, subnet, primaryDNS, secondaryDNS);**_

Not seen in any example...
Because the eztime needs a secondaryDNS I was running into another problem.

Now it works :
**_18:21:40.976 -> [hostByName] request IP for: pool.ntp.org
18:21:40.976 -> [hostByName] Host: pool.ntp.org IP: 66.228.58.20
18:21:41.120 -> [hostByName] request IP for: timezoned.rop.nl
18:21:41.163 -> [hostByName] Host: timezoned.rop.nl IP: 84.200.21.154_**

Thx a lot !

Erik

Georgie9117, 2023-08-03:
> WOW ! I forgot to extend the config call ...
>
> _**WiFi.config(ip, gateway, subnet, primaryDNS, secondaryDNS);**_
>
> Not seen in any example... Because the eztime needs a secondaryDNS I was running into [...]

Hi!

This solved my problem too. ezTime needs primary and secondary DNS the works fine in case when useing WiFi.Config() function. In my sketch it looks: `WiFi.config(IPAddress(192, 168, 1, 130), IPAddress(192, 168, 1, 1), IPAddress(255, 255, 255, 0), IPAddress(8, 8, 8, 8), IPAddress(8, 8, 4, 4));`

And after this for me started working the NTC library (_NTPClient.h_) as well.

G.