ropg/ezTime #53: Problem with using minuteChanged and secondChanged together

aussieW, 2019-06-22:
Hi

I am writing a program to display he time on an  LED matrix. I used secondChanged to blink a colon between the hours and minutes and also manually tracked when the minute changed in order to update the time on the display. That worked fine.

I figured I could simplify things by using minuteChanged to trigger a screen refresh rather than tracking when the minute changed. However, when I use both minuteChanged and secondChanged together I find that secondChanged is being triggered approx 100,000 times per second rather than just once per second.

My questions are: 
1. should the combination of these two functions work as I expect, and there is a bug occurring?
2. these two functions not designed to work together in the same program?

Thanks
aussieW

ropg, 2019-06-22:
You have to read the time using one of ezTime's functions (such as `now()`) for it to update `_last_read_t`, which is used internally to determine whether the seconds or minutes have changed since the last time the time was read. So just calling the function and then not reading the time when it returns true keeps it returning true. Just read the time with `now()` every time it has changed and you're fine.

closed 2019-06-22 by ropg

aussieW, 2019-06-22:
Thanks. Works great now.