ropg/ezTime #58: defines get redefined in PlatformIO

koter84, 2019-10-05:
I'm using the library with PlatformIO, one of the advantages is that when there is a new version of a library it's automatically updated.

When i want to change the EZTIME_LANGUAGE define (or any other for that matter) i need to change it in ezTime.h but when the library is updated this is undone by the update.
If i change it before importing ezTime.h in my main program it redefines it directly afterwards.

I think the only solution is to wrap defines in ezTime.h with #ifndef so they only get defined when they weren't already, but maybe there is a better solution?

ropg, 2019-10-06:
It's worse than that: the libraries get compiled separately from the main program, and only during linking are they combined. What that means is that you cannot set the define in the main program. I see no other solution than to change it every time the library updates. Yes this sucks.

koter84, 2019-10-06:
Yeah, i noticed when i was playing around with it...

But what i find really strange is that it gives an warning saying that the value gets redefined, while it actually doesn't do that since it's not defined in the library... ?!?

the only "workaround" i found is with build flags from the platformio.ini file ( which is in PR #60 )

closed 2019-10-29 by ropg