ropg/ezTime #64: Add CA (Catalan) language

clydde, 2019-10-27:
`
String monthStr(const uint8_t month) {
		switch(month) {
			case 1: return  F("gener");
			case 2: return  F("febrer");		
			case 3: return  F("març");
			case 4: return  F("abril");
			case 5: return  F("maig");
			case 6: return  F("juny");
			case 7: return  F("juliol");
			case 8: return  F("agost");
			case 9: return  F("setembre");
			case 10: return F("octubre");
			case 11: return F("novembre");
			case 12: return F("desembre");
		}
		return "";
	}

	String monthShortStr(const uint8_t month) { return monthStr(month).substring(0,3); }

	String dayStr(const uint8_t day) {
		switch(day) {
			case 1: return F("diumenge");
			case 2: return F("dilluns");
			case 3: return F("dimarts");		
			case 4: return F("dimecres");
			case 5: return F("dijous");
			case 6: return F("divendres");
			case 7: return F("dissabte");
		}
		return "";
	}

	String dayShortStr(const uint8_t day) {
		switch(day) {
			case 1: return F("Dg");
			case 2: return F("Dl");
			case 3: return F("Dt");		
			case 4: return F("Dc");
			case 5: return F("Dj");
			case 6: return F("Dv");
			case 7: return F("Ds");
		}
		return "";	
	}

`

ropg, 2019-10-27:
Can you make this a pull request? That way you also get proper credit in github...

ropg, 2019-10-27:
( Independència ! )

clydde, 2019-10-27:
> (¡Independència!)

;)

closed 2019-10-29 by ropg