ropg/ezTime #67: How can I use my local NTP Server with Internet connection?

x0tester0x, 2019-10-28:
I have a local NTP Server with Internet connection in my Network, The ESP8266, without Internet connection, is on the same Network.
How can the ESP8266 get the right Timezone from the local NTP Server?

ropg, 2019-10-28:
> I have a local NTP Server with Internet connection in my Network, The ESP8266, without Internet connection, is on the same Network.
> How can the ESP8266 get the right Timezone from the local NTP Server?

The timezone information does not come from the NTP server but from my own server (timezoned.rop.nl)

It's probably best if you find your timezone's POSIX information (the information that says what the timezone is called, what the offset to UTC is, and when the Daylight Savigs time starts and stops) and then use this information to set the timezone. If you say what timezone you are in I can give you the string.

x0tester0x, 2019-10-28:
I'm in UTC+1.
So I just have to set the Timezone what I will get from you and then I should get the right time always from my own NTP Server?

x0tester0x, 2019-10-28:
I'm in Italy.

ropg, 2019-10-28:
In that case, try something like:

```
Timezone Italy;
Italy.setPosix("CET-1CEST,M3.5.0,M10.5.0/3");
Serial.println(Italy.dateTime());
```

This way your timezone has the correct name (CET or CEST) and switches to DST at the right date (last Sundays of March and October). Remember to change this if and when DST is abolished in a few years though...

closed 2019-10-28 by ropg

x0tester0x, 2019-10-28:
But how can I retirieve then the Time from my local NTP Server after I set the Timezone?

ropg, 2019-10-28:
https://github.com/ropg/ezTime#setserver-and-setinterval

`void setServer(String ntp_server = NTP_SERVER);`

`void setInterval(uint16_t seconds = 0);`

>By default, ezTime is set to poll pool.ntp.org about every 30 minutes. These defaults should work for most people, but you can change them by specifying a new server with setServer or a new interval (in seconds) with setInterval. If you call setInterval with an interval of 0 seconds or call it as setInterval(), no more NTP queries will be made.

x0tester0x, 2019-10-28:
Now it works, but how can I get the local Time and not the UTC Time?