ropg/ezTime #69: bug in dateTime(now(), ...)

johnslemmer, 2019-11-12:
```c
	Timezone localTime;
	localTime.setDefault();
  localTime.setLocation("America/Los_Angeles");
  Serial.print("Current time: ");
  Serial.println(dateTime(ISO8601));
  Serial.print("Current time now(): ");
  Serial.println(dateTime(now(), ISO8601));
```

results in the following output:

```
Current time: 2019-11-11T17:57:26-0800
Current time now(): 2019-11-12T01:57:26-0800
```

I believe this is a bug with `dateTime()`.  The docs make it seem like these two calls to `dateTime` should be equivalent (unless I'm reading the docs incorrectly).

underwoodblog, 2019-11-12:
Doesn't need `localTime.setLocation("America/Los_Angeles"); `to be before` localTime.setDefault();`?

I've noticed that putting this in setup, doesn't always work, even if Timezone is global, but in loop it does.

johnslemmer, 2019-11-12:
@underwoodblog 

> Doesn't need `localTime.setLocation("America/Los_Angeles"); `to be before` localTime.setDefault();`?

No it doesn't.  `setDefault` just sets a certain `Timezone` instance to the default.  You can set/change the location as much as you want after that.  At least that is the behavior I have seen in my testing.

> I've noticed that putting this in setup, doesn't always work, even if Timezone is global, but in loop it does.

I have edited my original post to more clear about what is happening.  But I haven't noticed any issue with how I'm doing it.

ropg, 2019-11-12:
That does look like unexpected behaviour indeed. I just did a round of fixes, but let me look at this when I make some time for ezTime again. At that point I should be doing a larger overhaul to address larger structural issues as well, as the way I now deal with time_t in timezones is simply wrong. (It was an attempt to stay as close as possible to earlier arduino time libs, it turned into a bit of a mess.)

vi7, 2019-12-03:
@johnslemmer thanks for the fix! @ropg I confirm the bug either, looking forward for the PR merging and new version release

closed 2019-12-23 by johnslemmer