ropg/ezTime #74: eeprom.h: No such file or directory

Levin-D, 2019-12-19:
EEPROM.h is not found when using a MKR GSM 1400 board.

ropg, 2019-12-19:
There are two cache mechanisms available for ezTime to use. Either NVS or EEPROM. I don't know all this much about this board, but if you comment out the line that says `#define EZTIME_CACHE_EEPROM` in `src/ezTime.h` you should be good to go.

vbandke, 2020-01-13:
Same error here.  If I edit eztime.h and move line 

4. #ifdef __cplusplus

down to line 29, I.e. just before the „ Warranty void“ line, everything’s works out perfectly.  I have no idea, however, why this would make s difference.....

With best regards

Volker Bandke

ropg, 2020-01-13:
But is this with the `#define EZTIME_CACHE_EEPROM` commented out?

vbandke, 2020-01-13:
No, nothing was commented out.  When  the #ifdef __cplusplus is moved downwards, then the EEPROM.h file gets included nicely, and the Timezone.cpp program compiles and runs nicely.

vbandke, 2020-01-13:
Ha, one more info - while I am experiencing the same symptoms, the error is actually  not quite the same: I am using a Heltec Lora ESP32 board.

vbandke, 2020-01-13:
A follow-up:  the same issue occurs when I build for an Arduino Uno, in other words, the issue seems to be independent of the board that is configured

With best regards

Volker Bandke

ropg, 2020-01-13:
At this point I'm just really curious why this pops up all of a sudden with multiple people, it hasn't been an issue before, when already significant numbers of people were using ezTime.

vbandke, 2020-01-13:
Well, I am flabbergasted as well.  Maybe it is an issue with PlatformIO, who knows?  I‘ll open an issue there as well

ropg, 2020-01-13:
Is everyone else seeing this issue also using PlatformIO ?

vbandke, 2020-01-13:
I opened an issue at Platformio:  [https://github.com/platformio/platformio-vscode-ide/issues/1482](https://github.com/platformio/platformio-vscode-ide/issues/1482)

BTW, no issue here when using the Arduino IDE

merlinschumacher, 2020-04-29:
I am seeing this issue too on an ESP8266. But moving `#ifdef __cplusplus` doesn't work for me. Commenting out `#define EZTIME_CACHE_EEPROM` fixes it.

markymarkjay, 2021-01-18:
Just to pile on here: I've been running this on the ESP32 with no problems, but when I tried compiling for ESP8266, I got the same issue.  Commenting out works for me as well.

nabber00, 2022-04-19:
I am seeing this same issue with MKR Wifi 1010. The Arduino MKR Family does not have EEPROM, therefore I do not have or need EEPROM.h.  If the EEPROM library is a dependency, it should be added to the ezTime metadata to pull it in automatically with the library manager in the Arduino IDE to avoid this error.

On a related note, some of the MKR series include the ATECC508A security chip, which can provide similar functionality to EEPROM so you may want to add support for that.  The minimum slot size is 36 bytes which should be about right for a timezone to use one slot. There are 16 available slots.