ropg/ezTime #79: Request: Add sunrise / sunset calculation?

MacSass, 2020-02-03:
Hello, 
I really like this library - it works great, and still compact.

I´m not sure if this is totally out of scope, but was wondering if you could consider to add calculation of sunrise / sunset for a given place in local time ... 

Best regards and thanks for the nice library - MacSass

ropg, 2020-02-03:
Glad you like it!

Hmmm, that sounds more like a library of its own: all it needs is time, the rest is more geo-coordinate calculations. Or it might be part of some existing geo-coordinate library already. But there's also services online (some even free) that will give that to you for any given city, usually in combination with a crude weather forecast.

I'll close this here, but keep it in mind just in case.

closed 2020-02-03 by ropg

doanerock, 2020-02-04:
I agree that sunrise/sunset it out os scope.  Too many variables.

MacSass, 2020-02-04:
Ok, I do agree - was not sure myself - guess I was just selfish as I found ezTime a great working lib ...