ropg/ezTime #80: Comparing DateTime

Rich-Hopkins, 2020-03-04:
Is there a way that you would recommend for comparing datetimes? For example, I want to know if a timestamp received from an API is more than 2 hours prior to current time. Both times would be in UTC so timezone and DST conversion is not necessary.

ropg, 2020-03-04:
> Is there a way that you would recommend for comparing datetimes? For example, I want to know if a timestamp received from an API is more than 2 hours prior to current time. Both times would be in UTC so timezone and DST conversion is not necessary.

Eh... Simply get the times in seconds since epoch and see if they're more or less than 7200 seconds apart?

Rop

Rich-Hopkins, 2020-03-04:
Umm... Yeah. Thanks.\

closed 2020-03-04 by Rich-Hopkins