ropg/ezTime #81: Program crashes when logging data

posaidi, 2020-03-09:
hello

at the moment im working on a project where i use the M5Stack to get data from a different board and save it in a cloud every few seconds

. but my program crashes everytime when i try to execute a function where i try to log the data on thingspeak.com while i'm in a menu or submenu.

the function im trying to execute:

//----------------------------------------------------------------------------------------------
int writeThingSpeak(){                                              //Log Data on ThingSpeak.com
//----------------------------------------------------------------------------------------------
        ThingSpeak.setField(1,state); 
        ThingSpeak.setField(2,volt);
        ThingSpeak.setField(3,uamp);
          leistung = ((volt * uamp)/1000000);
        ThingSpeak.setField(4,leistung);
        ThingSpeak.setField(5,t_gas);
        ThingSpeak.setField(6,t_box);
        ThingSpeak.setField(7,vers_m);
        ThingSpeak.setField(8,vers_s);    
        ThingSpeak.writeFields(myChannelNumber, myWriteAPIKey); 
}

ropg, 2020-03-09:
Are you using M5ez? If so, this belongs there and not with ezTime, right?

posaidi, 2020-03-09:
yes, im using M5ez. sorry, my bad

ropg, 2020-03-09:
OK, closing here. Please refile in M5ez...

closed 2020-03-09 by ropg