ropg/ezTime #83: Several Question for implementation.

khawajamechatronics, 2020-03-28:
Hi,

I hope you are doing well. I am using this awesome library in a project. I have several queries. Instead of creating Issue of each query I prefer here as a thread.

I am not sure if there is any other platform where @ropg discuss issues or queries. 

1- local_or_utc = UTC_TIME; //Does this means UTC 0 time or GMT +0 time got from NTP? or set locally?
local_or_utc = LOCAL; // Whats the difference here? Does it is referenced to any time zone set. or just Arduino Time without NTP?

Thanks

khawajamechatronics, 2020-03-28:
Using myTZ.setDefault();
Does it affects UTC_TIME?
Does it affects LOCAL_TIME?

khawajamechatronics, 2020-03-28:
After going through several open issues, it seems Timezone is not fully working. Only UTC time is working. 

As per SetEvent function it seems it only using UTC time. 
`for (uint8_t n = 0; n < MAX_EVENTS; n++) {
			if (_events[n].function && nowUTC(false) >= _events[n].time) {
				debug(F("Running event (#")); debug(n + 1); debug(F(") set for ")); debugln(UTC.dateTime(_events[n].time));
				void (*tmp)() = _events[n].function;
				_events[n] = { 0, NULL };		// reset the event
				(tmp)();						// execute the function
			}
		}`

ropg, 2020-03-28:
Hi there,

I'm not sure I understand what you mean with the first question. NTP time is always in UTC. ("UTC 0 time or GMT +0 time got from NTP" ?), one has to add or subtract to get to data in any local timezone, which is what ezTime should do for you. If you create a timezone and set it to "Asia/Karachi", it should get the Posix string of `PKT-5` from the server, which means you are 5 hrs away from UTC.

`setEvent` can be supplied the time in local time or in UTC, as with most other functions in ezTime. Internally it uses UTC.

I am quite busy working on [this website](https://covid-at-home.info), so replies may be slow sometimes, but I'll try to answer if you provide me with clear simple three-line code segments that are not working for you.

khawajamechatronics, 2020-03-28:
Thanks for response. I am still exploring the library so apologize if I am unable to explain things well.

myTZ.setLocation(F("Pacific/Auckland"));
myTZ.setDefault();

It will set default timezone.

But when I call 
Serial.print(hour(TIME_NOW,LOCAL_TIME));
or
Serial.print(hour(TIME_NOW,UTC_TIME));
it prints same results. in short UTC_TIME or LOCAL_TIME has no affect on several functions I have tested.

May be issue #10 ?

In that case I have to play with library very carefully.

khawajamechatronics, 2020-03-28:
Let me give you example with functions
In Sketch
myTZ.setLocation(F("Asia/Karachi"));
myTZ.setDefault();
Serial.println(now());
In Source Code
time_t now() { return  (defaultTZ->now()); }
So above results should have been printed UTC Seconds in ASIA/Karachi Timezone
But it printed in GMT 0 time.

ropg, 2020-03-29:
The first example does print two different hours for me. Are you sure your time synchronizes?

Can you this sketch and post the output?

```
#include <ezTime.h>
#include <WiFi.h>

void setup() {

	setDebug(INFO);
	Serial.begin(115200);
	WiFi.begin("your-ssid", "your-password");
	waitForSync();

	Serial.println("UTC: " + UTC.dateTime());
	
	Timezone NewZealand;
	NewZealand.setLocation("Pacific/Auckland");
	Serial.println("New Zealand time: " + NewZealand.dateTime());
}

void loop() { }
```

khawajamechatronics, 2020-03-29:
Yes this does print 2 different time. 

I am talking about 

#include <ezTime.h>
#include <WiFi.h>

void setup() {

	setDebug(INFO);
	Serial.begin(115200);
	WiFi.begin("your-ssid", "your-password");
	waitForSync();

	Serial.println("UTC: " + UTC.dateTime());
	
	Timezone NewZealand;
	NewZealand.setLocation("Pacific/Auckland");
	Serial.println("New Zealand time: " + NewZealand.dateTime());

Serial.println(now());
NewZealand.setDefault();
Serial.println(now());
}
Should above 2 prints of now() be same or different?
Considering defaults are changed using setDefault?

khawajamechatronics, 2020-04-02:
@ropg I know you busy. I hope you find time to review this.

Thanks

ropg, 2020-04-02:
I guess it should print two different numbers. Doesn't it?

khawajamechatronics, 2020-04-02:
@ropg Apologize, I wanted to reference to different variable, yes now() prints different time after setDefault.

Can you check this code as its hitting exception when digitalClockDisplay is called.

Exception doesn't happen when I print hour,min,seconds   or   day month year, seperate but I combine both set as in code below its hitting exception. not sure why.

    #include <ezTime.h>
    #include <ESP8266WiFi.h>
    long lastTimeMillis;
    Timezone myTZ;
    void setup() {
    Serial.begin(115200);
    while (!Serial) {
    ;  // wait for Serial port to connect. Needed for native USB port only
    }
    WiFi.begin("Slow Connection_4G", "chumchum44");
    // Uncomment the line below to see what it does behind the scenes
    setDebug(DEBUG);
    waitForSync();
    Serial.println("UTC: " + UTC.dateTime());
    Timezone NewZealand;
    NewZealand.setLocation("Asia/Karachi");
    Serial.println("Pakistan time: " + NewZealand.dateTime());
    Serial.println(now());
    NewZealand.setDefault();
    Serial.println(now());
    // Serial.println(now(),UTC_TIME);
    Serial.println(millis());
    }
    void loop() {
    digitalClockDisplay();
    events();
    }
    void digitalClockDisplay() {
    if (millis() - lastTimeMillis > 5000)
    {
    // digital clock display of the time
    Serial.print(hour());
    // Serial.print(hour(TIME_NOW,LOCAL_TIME));
    printDigits(minute());
    printDigits(second());
    Serial.print(" ");
    Serial.print(day());
    Serial.print(" ");
    Serial.print(month());
    Serial.print(" ");
    Serial.print(year());
    Serial.println();
    lastTimeMillis = millis();
    Serial.println(now());
    }
    }
    void printDigits(int digits) {
    // utility function for digital clock display: prints preceding colon and leading 0
    Serial.print(":");
    if (digits < 10)
    Serial.print('0');
    Serial.print(digits);
    }
`

khawajamechatronics, 2020-04-04:
Exception Decoder
[image]

khawajamechatronics, 2020-04-04:
Library is very useful despite of minor issues or misunderstanding. Thanks for your work @ropg

khawajamechatronics, 2020-04-04:
@ropg 
2 important questions
- How to makeTime in specific timezone, currently it seems it maketime in UTC time zone

- Also in uint8_t Timezone::setEvent
debugln(defaultTZ->dateTime(t)); debugln(UTC.dateTime(t));

prints same time when I have different default time zone, 

Am I missing something or is it a bug?

khawajamechatronics, 2020-04-04:
It seems the issue is debugln(defaultTZ->dateTime(t)); as it shows incorrect time and it confuses, can you let me know how to show correct time.