ropg/ezTime #85: NodeMcu / Arduino Timekeeping with ezTime

FabRo57, 2020-04-01:
Hello , this is a follow-up of a previuos issue regarding my sketch , which works like a charm calling every time the NTP server service : my target is getting only once in a while NTP time and making it possible in some way using an on-board time .
Though much research I concluded I have no skill to apply the traces I found , and I am pretty sure it could be possible with ezTime lib ...
Here is the very extract of my program : 

void setup()
{
// ...extract of my sketch running on NodeMCU ESP-12E
// ...it uses also  websockets , SDcard read/write, temperature readings  and tickerscheduler ..
// ...
// ...

  Serial.begin(115200);

  //Serial.setDebugOutput(true);              

  Serial.println();
  Serial.println();
  Serial.println();

  for(uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] BOOT WAIT %d...\r\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {

    Serial.print(".");
    delay(100);
  }

  if (mdns.begin("espWebSock", WiFi.localIP())) {
    Serial.println("MDNS responder started");
    mdns.addService("http", "tcp", 1234);          
    mdns.addService("ws", "tcp", 1235);            
  }
  else {
    Serial.println("MDNS.begin failed");
  }
  Serial.print("Connect to http://espWebSock.local or http://");
  Serial.println(WiFi.localIP());

  server.on("/", handleRoot);
  server.onNotFound(handleNotFound);

  server.begin();

  webSocket.begin();
  webSocket.onEvent(webSocketEvent);

     setServer("it.pool.ntp.org");               //                 
     myTz.setLocation("Europe/Rome");		 //
     myTz.setDefault();			        //	
     
     setDebug(DEBUG);   			//  	
     waitForSync();                                       
     Orario=myTz.dateTime("H\:i\:s\dMy");       //   
  } 	

//***************************

void Enercheck() {

PowState=digitalRead(Ener);  // Pin check

	if ((PowState==false)&&(flag==false)) {
		NTP_aggiorna();
		Messaggio=("Assenza Energia: "+ Ore);
		pushover_button.setMessage(Messaggio);
		Serial.println(pushover_button.send());
		flag=true;
		webSocket.broadcastTXT(RETEOFF);  }

	if ((PowState==true)&&(flag==true)) {
		NTP_aggiorna();
		Messaggio=("Ritorno Energia: "+ Ore);
		pushover_button.setMessage(Messaggio);
		Serial.println(pushover_button.send());
		flag=false;
		webSocket.broadcastTXT(RETEON);  }
  }

//***************************

void loop()
{
     webSocket.loop(); 			

     server.handleClient();
       
     ts.update();        	//Tickerscheduler

     Enercheck();     

     events();
}

void NTP_aggiorna() {

      	waitForSync();
      	Orario=myTz.dateTime("H\:i\:s\dMy");
        Ore=Orario.substring(0,8);	

  }

ropg, 2020-04-01:
I'm sorry but have no idea what the question or problem is here... Can you be more clear?

FabRo57, 2020-04-01:
Thanks for reply and sorry for my confusion : my question is : is it possible on nodemcu keep current time correcting it just once in a while, using ezTime? If so,  how could it be? Thanks.  FabioInviato da smartphone Samsung Galaxy.
-------- Messaggio originale --------Da: Rop Gonggrijp <notifications@github.com> Data: 01/04/20  18:09  (GMT+01:00) A: ropg/ezTime <ezTime@noreply.github.com> Cc: FabRo57 <fabrock0357@gmail.com>, Author <author@noreply.github.com> Oggetto: Re: [ropg/ezTime] NodeMcu / Arduino Timekeeping with ezTime (#85) 
I'm sorry but have no idea what the question or problem is here... Can you be more clear?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ropg/ezTime/issues/85#issuecomment-607342371",
"url": "https://github.com/ropg/ezTime/issues/85#issuecomment-607342371",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

FabRo57, 2020-04-01:
I guess I casted even more confusion . Let's try again :+1: 
1 - ezTime works correctly each time it needs access to NTP server .
2 - sometimes 'too close' access give me wrong results ( typically the second one )
3 - I wondered if it was possible have 'on board time' , like in time.h library I red somewhere , but using ezTime , which deals with DST and TZ easily .
4 - finally I need a trace on how get that : at setup  ask NTP time , put it in a sort of routine which gives me the current (correct!!) time without any furter ntp server connections , apart from the scheduled updates .
.. hope it's a bit clearer now , sorry again for both confusion and unfluent English .

Fabio R.

khawajamechatronics, 2020-04-08:
I think EZtime sync onboard Arduino time with NTP Server in setInterval.

You can access onboard time. There are millis funcion as well as now() function which are used in time.h library, 

If its not what you meant then please explain.