ropg/ezTime #87: enum NONE SSDP

totalretribution, 2020-04-19:
If i use ezTime with ESP8266SSDP.h i get an error "error: redeclaration of 'NONE'".
Ideally you should not use enums with such a common name in the global namespace.

ropg, 2020-04-19:
True... I am actually quite a new programmer: ezTime is my second project in C++ (M5ez was the first)

But if you uncomment the define for `EZTIME_EZT_NAMESPACE` in the header file, all of ezTime moves to the `ezt` namespace for easier integration into larger projects. Please let me know if that works for you.

totalretribution, 2020-04-21:
Sadly that just gives me lots of other errors and still includes the redeclaration error. 
Your programming skills are better than mine since I am an amateur at best.

`.pio\libdeps\nodemcuv2\ezTime_ID5751\src\ezTime.cpp: In member function 'bool Timezone::setLocation(String)':
.pio\libdeps\nodemcuv2\ezTime_ID5751\src\ezTime.cpp:832:23: error: 'error' was not declared in this scope
    error (SERVER_ERROR);
                       ^
.pio\libdeps\nodemcuv2\ezTime_ID5751\src\ezTime.cpp:832:23: note: suggested alternative:
.pio\libdeps\nodemcuv2\ezTime_ID5751\src\ezTime.cpp:135:12: note:   'ezt::error'
  ezError_t error(const bool reset /* = false */) {
            ^
.pio\libdeps\nodemcuv2\ezTime_ID5751\src\ezTime.cpp:847:24: error: 'error' was not declared in this scope
   error (DATA_NOT_FOUND);
                        ^
.pio\libdeps\nodemcuv2\ezTime_ID5751\src\ezTime.cpp:847:24: note: suggested alternative:
.pio\libdeps\nodemcuv2\ezTime_ID5751\src\ezTime.cpp:135:12: note:   'ezt::error'
  ezError_t error(const bool reset /* = false */) {
            ^
In file included from ../src/main_simple.ino:34:0:
.pio\libdeps\nodemcuv2\ezTime_ID5751\src/ezTime.h:69:2: error: redeclaration of 'NONE'
  NONE, 
  ^
In file included from ../src/main_simple.ino:29:0:
.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266SSDP/ESP8266SSDP.h:51:3: note: previous declaration 'ssdp_method_t NONE'
   NONE,`

ropg, 2020-04-21:
Hrmz... I'll have to do some testing with this at some point. I also see you're using platformio and not the Arduino environment?

totalretribution, 2020-04-21:
Im using Platformio, however its an Arduino project and as such should be the same as using the standard Arduino IDE.

gilsonmandalogo, 2023-12-26:
Do we have some update on this? I have the exact same error if I define `EZTIME_EZT_NAMESPACE`.