ropg/ezTime #96: Memory leak in NTP sync logging?

smuellener, 2020-08-04:
I run ezTime on an ESP32.  NTP sync is set up in ezTime with an interval of 60 seconds. I see info log messages ("spot on", sometimes "1ms fast")

- ESP.getFreeHeap() returns 100 bytes less after every sync with NTP server (once every minute)
- If I set the NTP interval of ezTime to 10 minutes: ESP.getFreeHeap() returns 100 bytes less after every NTP sync (once every 10 minutes!)
- Defining EZTIME_MAX_DEBUGLEVEL_NONE does result in constant ESP.getFreeHeap()

Is there a memory leak in the logging of ezTime?

ropg, 2020-08-04:
Hmmm, definitely something to look at. The fact that it's not there with `EZTIME_MAX_DEBUGLEVEL_NONE` is interesting. How about `..._ERROR` and `..._INFO` ?

Really all that does is call or not call the `print` or `println` method of the serial port. (See lines 29-57 of https://github.com/ropg/ezTime/blob/master/src/ezTime.cpp) It seems odd to me that that would cause a leak, but maybe I am missing something.

By all means feel free to investigate further...

smuellener, 2020-08-05:
Interesting, I decreased the interval and the reduction in the Heap is only during the first approx. 10 syncs, then it stays stable. And it also happens without logging activated. As we only talk about 10*100 bytes I can live with that. It would still be interesting to know where this comes from. To clearify: calling ESP.getFreeHeap() can not be the issue as if I call it more often, the decrease in the free heap space does not happen more quickly but stays in sync with the NTP sync of ezTime.

Another effect that I came across during testing: After syncing every 5 seconds for the testing above, I sometimes got:

`Querying 172.20.0.1 ... ERROR: Timeout`

from the NTP server (PFSense). Not much of a problem. But I also got the follwing sometimes (and always before the timeout above):

`Received time: Thursday, 07-Feb-36 06:28:16.001 UTC (internal clock was 258154746 ms slow)`

That is a problem of course as it sets the time of the device to 2036. I will further investigate. Please let me know if you have an explanation for this.

smuellener, 2020-08-11:
Will close this as limited only to approx. 10 syncs and hence unproblematic

closed 2020-08-11 by smuellener