ropg/ezTime #97: setEvent not firing

stephensaid, 2020-08-09:
I have this sketch that should fire a setEvent but I'm failing to see what I'm doing wrong.
I do not have wifi connection so am setting the time using compileTime().

```
#include <ezTime.h>

void getTime();
void displayTime();

int w = 10;
String curTime;

void setup() {
  Serial.begin(115200);
  delay(500);

  getTime();

  curTime = dateTime("H:i:s A") + "   ";
  Serial.print("dateTime: ");
  Serial.println(curTime);
  
  Serial.print("now(): ");
  Serial.println(now());

  Serial.print("now() formatted: ");
  Serial.print(hour(now()));
  Serial.print(":");
  Serial.print(minute(now()));
  Serial.print(":");
  Serial.print(second(now()));
  Serial.println();
  Serial.println();

  setEvent(displayTime, now() + w);
}

void loop() {
  events();
}

void displayTime() {
  curTime = dateTime("h:i A") + "   ";
  Serial.println(curTime);
  Serial.println();

  setEvent(displayTime, now() + w);
}

void getTime() {
  setTime(compileTime());
}
```

This is the output I get:

```
dateTime: 19:54:23 PM   
now(): 1597002863
now() formatted: 19:54:23

```

Any help is greatly appreciated.

stephensaid, 2020-08-10:
I continued to investigate but am getting nowhere!
Although the event is being set, it never fires.

```
#include <ezTime.h>

void displayTime();

int w = 10;
String curTime;

void setup() {
  Serial.begin(115200);
  while (!Serial) {};

  Serial.println("-- ezTime Test Sketch --");

  setDebug(DEBUG);
  setTime(compileTime());
  setInterval(0);

  curTime = dateTime("l, d-M-y H:i:s.v T") + "   ";
  Serial.print("dateTime:       ");
  Serial.println(curTime);

  setEvent(displayTime, now() + w);
  delay(10);
}

void loop() {
  events();
}

void displayTime() {
  Serial.println("displaying time... ");
  setEvent(displayTime, now() + w);
}
```

This is the output I'm getting from the above:
```
-- ezTime Test Sketch --

ezTime debug level set to DEBUG
dateTime:       Monday, 10-Aug-20 20:09:00.000 UTC   
Set event (#1) to trigger on: Monday, 10-Aug-2020 20:09:10 UTC
```

I really need some help, please.

ropg, 2020-08-10:
After staring at it for a while, I cannot see anything obviously wrong with your code. Don't have an Arduino handy right now, but could you print the value of `compileTime()` as well as `nowUTC(false)` after printing the time?

I suspect this has something to do with not syncing with NTP, but cannot find anything obvious in the code.

stephensaid, 2020-08-10:
Can't use ```nowUTC(false);``` it seems!
It does not compile: ```'nowUTC' was not declared in this scope```

This is what I've got now:
```
#include <ezTime.h>

void displayTime();

int w = 10;
String curTime;

void setup() {
  Serial.begin(115200);
  while (!Serial) {};

  Serial.println("-- ezTime Test Sketch --");

  setDebug(DEBUG);
  setTime(compileTime());

  setInterval(0);

  curTime = dateTime("l, d-M-y H:i:s.v T") + "   ";
  Serial.print("dateTime:       ");
  Serial.println(curTime);

  Serial.print("compileTime: ");
  Serial.println (compileTime());

  Serial.print("now: ");
  Serial.println (now());

//  nowUTC(false);

  curTime = dateTime("l, d-M-y H:i:s.v T") + "   ";
  Serial.print("dateTime:       ");
  Serial.println(curTime);

  setEvent(displayTime, now() + w);
  delay(10);
}

void loop() {
  events();
}

void displayTime() {
  Serial.println("displaying time... ");
  setEvent(displayTime, now() + w);
}
```

And this is the output:
```
-- ezTime Test Sketch --

ezTime debug level set to DEBUG
dateTime:       Monday, 10-Aug-20 23:41:11.000 UTC   
compileTime: 1597102871
now: 1597102871
dateTime:       Monday, 10-Aug-20 23:41:11.002 UTC   
Set event (#1) to trigger on: Monday, 10-Aug-2020 23:41:21 UTC
```

Thanks for looking into this.

ropg, 2020-08-11:
Eh, sorry, that's the private function internally. I meant `UTC.now()`

stephensaid, 2020-08-11:
Here's the updated sketch.
```
#include <ezTime.h>

int w = 10;
String curTime;

void setup() {
  Serial.begin(115200);
  while (!Serial) {};

  Serial.println("-- ezTime Test Sketch --");

  setDebug(DEBUG);
  setTime(compileTime());

  setInterval(0);

  curTime = dateTime("l, d-M-y H:i:s.v T") + "   ";
  Serial.print("dateTime:       ");
  Serial.println(curTime);

  Serial.print("compileTime: ");
  Serial.println (compileTime());

  Serial.print("now:         ");
  Serial.println (now());

  Serial.print("UTC.now:     ");
  Serial.println(UTC.now());

  curTime = dateTime("l, d-M-y H:i:s.v T") + "   ";
  Serial.print("dateTime:    ");
  Serial.println(curTime);

  setEvent(displayTime, now() + w);
  delay(10);
}

void loop() {
  events();
}

void displayTime() {
  Serial.println("displaying time... ");
  setEvent(displayTime, now() + w);
}

```
Output:
```
-- ezTime Test Sketch --

ezTime debug level set to DEBUG
dateTime:       Tuesday, 11-Aug-20 19:23:38.000 UTC   
compileTime: 1597173818
now:         1597173818
UTC.now:     1597173818
dateTime:    Tuesday, 11-Aug-20 19:23:38.005 UTC   
Set event (#1) to trigger on: Tuesday, 11-Aug-2020 19:23:48 UTC
```

stephensaid, 2020-08-13:
Hi
Wondering if you had the time to have another look, please?

Thanks

nouc23, 2021-12-31:
Hi @stephensaid.
I occur similar problem. In library, during first `events()` call all events are cleared. You need to call `events()` in setup, before you call `setEvent`