ropg/ezTime #98: DST not properly calculating

sparer, 2020-08-12:
I have the following code:

	setInterval(60);
	setDebug(INFO);
	waitForSync();
	Timezone tz;
	tz.setLocation("Europe/London");
	String posix = tz.getPosix();
	Serial.println("POSIX1: "+posix);
	Serial.println("IsDST: " + String(tz.isDST(1521939599, UTC_TIME))); //Sunday, March 25, 2018 0:59:59
	Serial.println("IsDST: " + String(tz.isDST(1521939600, UTC_TIME))); //1 second later

... and the serial output:
	Received time: Wednesday, 12-Aug-20 11:15:43.262 UTC
	Time is in sync
	Timezone lookup for: Europe/London ... (round-trip 28 ms)  success.
	  Olsen: Europe/London
	  Posix: GMT0BST,M3.5.0/1,M10.5.0
	POSIX1: GMT0BST,M3.5.0/1,M10.5.0
	IsDST: 0
	IsDST: 1

This is all good however, the following year DST occurs 31-03-2019 at 1:00:00. See: https://www.gov.uk/when-do-the-clocks-change
So using:
	Serial.println("IsDST: " + String(tz.isDST(1553993999, UTC_TIME))); //Sunday, March 31, 2019 0:59:59
	Serial.println("IsDST: " + String(tz.isDST(1553994000, UTC_TIME))); //1 second later

...  the serial output:
	Received time: Wednesday, 12-Aug-20 11:21:44.092 UTC
	Time is in sync
	Timezone lookup for: Europe/London ... (round-trip 28 ms)  success.
 	 Olsen: Europe/London
  	Posix: GMT0BST,M3.5.0/1,M10.5.0
	POSIX1: GMT0BST,M3.5.0/1,M10.5.0
	IsDST: 1
	IsDST: 1

I think that DST is incorrectly calculated against the POSIX. It's the same for 2020 and 2021

sparer, 2020-08-12:
Remade project and reinstalled library and now appears to be working...strange.

closed 2020-08-12 by sparer

ropg, 2020-08-12:
Glad it works now. Interesting failure, but no clue what could have happened there...