ropg/ezTime #101: Longest posix string?

csgregg, 2020-10-09:
I am wanting to store the posix for the current timezone and need to know the longest possible posix string length. Can't see anything in the source or server that might shed light on it. 

Anyone know?

Thanks!

csgregg, 2020-10-09:
A bit of deduction ...

MAX_CACHE_PAYLOAD is olsen + " " + posix
I think the olsen is 29 (14 + / + 14)
So posix might be 32

Can anyone confirm?

Docteh, 2021-02-27:
Hi, I poked around in /usr/share/zoneinfo and found three that are longer than 32

* Iran (34 chars): `<+0330>-3:30<+0430>,J79/24,J263/24`
* Australia/LHI (36 chars) : `<+1030>-10:30<+11>-11,M10.1.0,M4.1.0`
Lord Howe Island
* NZ-CHAT (44 chars): `<+1245>-12:45<+1345>,M9.5.0/2:45,M4.1.0/3:45`
Chatham Islands