ropg/ezTime #104: setLocation() and GeoIP not working anymore?

eyecee, 2021-01-07:
Hi - I've been using the ezTime library for 3 months and it's been working really well where I've been getting the local timezone automatically using the setLocation() function which auto-detected my local timezone via GeoIP. Occasionally it would time out so I put in a backup to define the local timezone and a fallback to default to UTC. there is the code:

`waitForSync();
  Timezone myTZ;
  
  if (myTZ.setLocation()) { 
    Serial.println(myTZ.dateTime());
  } else if (myTZ.setLocation(F("DE"))) {
    Serial.println(myTZ.dateTime());
  } else {
    Serial.println(UTC.dateTime());
  }`

This was working perfectly for a few months but then 2 days ago for some unknown reason, the auto-location stopped working and only provided the UTC time. Can I check to see if the "timezoned.rop.nl" server is up or if there is some other issue?

merlinschumacher, 2021-01-16:
I have exactly the same issue.

merlinschumacher, 2021-01-17:
After checking further, I found the issue being with the ntp service. The sync with `pool.ntp.org` seems to be failing. using my wifi router as ntp server fixes this.

eyecee, 2021-01-17:
@merlinschumacher - thanks for the confirmation of the issue and a proposed solution. May I ask; how can I replicate your solution and set up my router as an NTP server? Is it an inherent function that I can enable in the router settings?

merlinschumacher, 2021-01-17:
Some routers offer this function, some don't. It highly depends on the model. But you can try any other timeserver. 
Here is a list of possible servers:
https://gist.github.com/mutin-sa/eea1c396b1e610a2da1e5550d94b0453

eyecee, 2021-01-17:
Thanks for your quick reply @merlinschumacher! No worries on the timeservers; from the eztime code side, have you used the 'setServer()' function to change the default NTP server  and are there any other lines of code that need to be defined to point ezTime to the redefined NTP server?

I would have thought that ezTime's hosted timezoned.rop.nl server was causing the issue by not pinging back the geoID from the IP address contacting the server. If I reroute to another timeserver, will they do a similar ping back?

Thanks in advance.

doanerock, 2021-01-17:
> The sync with `pool.ntp.org` seems to be failing. using my wifi router as ntp server fixes this.

I don't understand as long as the NTP server is responding it should not matter if the NTP server is on a local network or on the internet.

merlinschumacher, 2021-01-19:
@doanerock It seems pool.ntp.org sometimes points to a server that doesn't reply properly or replys in a manner that eztime won't handle properly. I just used my router for convenience, local or remote is not the issue.

@eyecee eztime uses two servers. any NTP server to get the current time in UTC and `timezoned.rop.nl` to get the current timezone data to create the local time.
`timezoned.rop.nl` presumably works fine (at least in my case), but using pool.ntp.org (which is an alias for a multitude of different NTP servers) makes the library fail for me.

doanerock, 2021-01-19:
@merlinschumacher  "It seems pool.ntp.org sometimes points to a server that doesn't reply properly or replys in a manner that eztime won't handle properly. I just used my router for convenience, local or remote is not the issue."  
I see how it is possible how a hand full of many NTP servers could be down at any one time.  If one NTP server does not respond in time would it be possible to have ezTime try the next server in the pool.ntp.org list?  Or wait for the DNS TTL to timeout and do new lookup for pool.ntp.org.

merlinschumacher, 2021-01-19:
I've used ` waitForSync(30);` in my code to give it a 30 sec timeout. Afterwards it just uses the unix epoch beginnging (January 1st, 1970, 00:00). I don't know if ezTime tries to resync the servers after failing once, but you certainly could trigger another call to `waitForSync()`.