ropg/ezTime #105: ezTime w/MKR 1010 or MKR1000

cgleon809, 2021-01-13:
Hello!
First of all congrats on a very well documented and thought out library! 
I am having trouble compiling ezTime for MKR 1010 (or MKR 1000) boards.  After fiddling with ezTime.h and making EZTIME_CACHE_NVS defined, while commenting EZTIME_CACHE_EEPROM I get an error indicating that the compiler could not find "preferences.h".

Can you please advise?

imranmoezkhan, 2021-01-14:
Hi @cgleon809 ,
I had this same issue when I started using the ezTime lib a couple of weeks ago with my MKR1010. If you search for "mkr1000" on the main page of this repo, you'll see this bug has been recorded:

_It doesn't have EEPROM.h or Preferences.h but some proprietary FlashStorage.h. So no cache for the moment. (Turn off both cache defines at the beginning of ezTime.h. I'll write it if the third person wants it._

So the workaround to use this is to turn off both caches and comment out #include <Preferences.h> in ezTime.cpp . Unfortunately its not possible to use the timezone caching on the mkr1000/1010 boards at the moment.

cgleon809, 2021-01-14:
Really(!) appreciate it!  Thanks!

closed 2021-01-27 by cgleon809

ArcaicoJ, 2021-05-02:
Thank you very much for the answers, I have the same problem when trying to compile for my Arduino MKR1000, I am not very good with the C ++ language, so how can you turn off those caches? , please