ropg/ezTime #106: Struggling to understand how to implement with M5.ez

andyrbarrow, 2021-01-15:
Hi Rop,
Just trying to reach that "ah ha" moment with ez.time and implementation with M5.ez. I'm using a GPS string to update time, which seems to work well as a one-shot thing or as part of a loop. Since I don't want to tie things up by updating the clock on every loop, I want to use events() as a way of kicking off a set event to update the time. 

What I'm struggling with is where events() goes. Since my loop() is essentially rendered unnecessary (and I guess unusable) by <menuname>.run(), where would would I place events()?

Thanks for a great couple of libraries! Really helpful in my smart boat projects.

ropg, 2021-01-15:
Some next version should have better support for external time sources, ezTime is really built around NTP.

That said: you could create an event that sets the time. If your GPS delivers a 1PPS signal, you could hook that to a pin and use the edge of that to exactly set the time, making it much more precise than just using the NMEA text string from the GPS.

As to the question at hand:

M5ez runs it for you in it's [code that updates the clock](https://github.com/M5ez/M5ez/blob/master/src/M5ez.cpp#L969), so it always runs as long as M5ez's [`ez.yield()`](https://github.com/M5ez/M5ez/blob/master/src/M5ez.cpp#L2189) is called, which includes all menus and other moments where M5ez is waiting for keys.

Please let me know if anything is still unclear.

andyrbarrow, 2021-01-15:
Right now, I just capture a SignalK key called navigation.datetime that is the GPS text string for current time, which is parsed in a function and settime() is called from that. That function works great if I call it on a regular basis. My question was about where and how to call that function as part of M5.ez. Accuracy isn't critical, but I can't just set the clock once as it drifts out pretty far over several hours.

Sounds like I need to put a setevent() call to my function in each menu? 
My loop() looks like this now:
`void loop() {
  ezMenu mainmenu("Hey Ya Info System");
  mainmenu.txtBig();
  mainmenu.addItem("Location", location_display);
  mainmenu.addItem("Wind", wind_display);
  mainmenu.addItem("Tanks", tank_display);
  mainmenu.addItem("Electrical", electrical_display);
  mainmenu.addItem("Settings", ez.settings.menu);
  mainmenu.upOnFirst("last|up");
  mainmenu.downOnLast("first|down");
  mainmenu.run();
}`
Is it something like placing setevent() somewhere in here for this particular menu?