ropg/ezTime #108: Timeout on NTP Servers ... any server

cgleon809, 2021-01-27:
Hi!
I tried running your code Ethernet.ino and I get a time out.for NTP server.  Further, I tried changing timeX.google.com, timeX.facebook.com, various ntp.pool.org and still getting timeout.

MkR is connected to Ethernet router, and I am getting an IP from DHCP.  

Any hints?

Pidbid, 2021-01-27:
There are some NTP servers for you,
time.apple.com,time.windows.com,time.nist.gov
hope this helps

cgleon809, 2021-01-27:
Thanks!
I actually dont think is a server issue.  I've tried many, including the
ones you indicated.  I just tried, time.nist.gov, and get same error.
Is it possible that a UDP port may be closed?  How can I find out?

Pidbid, 2021-01-27:
I got the same problem today,But my Arduino borad often runs twice and finally succeeds. Here's some code that might help,NTP service based on port 123, you should run "nc -vuz IP 123"

cgleon809, 2021-01-27:
I can't thank you enough!  I did as you suggested, the problem is still
there.    Do you have further recommendations that I can try, I am really
desperate!
I am running the Ethernet.ino example provided in ezTime library.

doanerock, 2021-01-27:
Start from the basics.  Can your device ping the servers?
https://github.com/BlakeFoster/Arduino-Ping

cgleon809, 2021-01-27:
Thanks, I'll check it out.
I've been using queryNTP function from ezTime library getting a time-out. I
will try your recommendation.

cgleon809, 2021-01-27:
Just tried!!
Can't get it to work, as I am getting the same issues as reported in
https://forum.arduino.cc/index.php?topic=573949.0.  Any suggestions?

jgrulich, 2021-04-04:
I'm using manual NTP update and have the same issue. Not all the time, but sometimes it report timeout. When I use my local server, it never happens. I've tried may different internet ntp servers, but doesn't care.
Even more critical issue is that it changes the time to some strange value in the future, included the date, just 5 minutes after the new and successful ntp update. Today it displayed "Monday, 23.5.2021, 05:10", see the attached report. "No network" on the report is due to WiFi is switched off just after the `updateNTP()`. Each `updateNTP()` is performed only when `WiFi.status() == WL_CONNECTED`. Please help.
[ezTime.txt](https://github.com/ropg/ezTime/files/6254518/ezTime.txt)

jgrulich, 2021-04-04:
In my case will help if the `updateNTP()` will return status flag. I was looking for some status indicator, but only `timeStatus() != timeSet` which is not triggered either. Any other valid status info indicating wrong update?