ropg/ezTime #122: Is timeserver not responding?

andersbjn, 2021-09-16:
Is there some error with the server?
It doesn't respond to requests since around 10:30 Stockholm time?

ropg, 2021-09-16:
> Is there some error with the server?
> It doesn't respond to requests since around 10:30 Stockholm time?

Correct, I'm upgrading... Sorry for the disruption...

andersbjn, 2021-09-16:
Ok!
Thank you for the superquick response!
When do you yhink it will be up again?
Best regards, Anders

ropg, 2021-09-16:
> Thank you for the superquick response!
> When do you yhink it will be up again?

It's up again.

andersbjn, 2021-09-16:
Supermany thanks!
Best regards, Anders

SergDfv, 2021-09-16:
https://github.com/ropg/ezTime/blob/7b3c8aa020be818ac149e0762543ac5e81ccfabe/src/ezTime.h#L153-L155

rvdbreemen, 2021-09-16:
@ropg can't resolve timezones anymore, so did your upgrade break stuff?

projectsun2, 2021-09-17:
Hello, The server retreive of TZ location fails

andersbjn, 2021-09-17:
Hi!
Do you have any update of what has happened and when it will work again?
Best regards,
Anders

TomasOq, 2021-09-17:
Hello, The server is currently not responding - do you know when it will be available again?
Thanks in advance.

dos1, 2021-09-17:
FWIW https://github.com/ropg/ezTime/pull/123 makes setting a custom server easier.

ropg, 2021-09-17:
Holy crap, the 2343 was a mistake on my end, and then I didn't check my computer again. I'm so, so sorry...

it should now be up at 2342. Please test.

Sorry...

(And yes, I think it would make sense to have a service like timezoned have more servers, but it's my own homebrew protocol instead of something standardized that has an RFC and all. Will work on putting it on the same box that hosts my backup nameserver so there's at least two.)

closed 2021-09-17 by ropg

SergDfv, 2021-09-17:
[image]
The port is still not open...

andersbjn, 2021-09-17:
Now server is responding with timezone definition on port 2342 so it seems ok!
Thanks!

projectsun2, 2021-09-17:
> 
> 
> [image]
> The port is still not open...

You can´t scan UDP! Because it´s a connectionless protocol.

ropg, 2021-09-18:
To check if it's working, use:

```bash
$ nc -u timezoned.rop.nl 2342
```

And then type any identifier and follow it with a Ctrl-D (and no return). You'll then see the timezone info right after what you typed:

```
londonOK Europe/London GMT0BST,M3.5.0/1,M10.5.0
```

TheGroundZero, 2023-07-03:
> And yes, I think it would make sense to have a service like timezoned have more servers

Did a crude attempt here, putting your code in a container image.

https://github.com/TheGroundZero/ezTimeContainer