ropg/ezTime #125: Server still not responding after upgrade

andersbjn, 2021-09-17:
Hi!
Sorry for creating a new issue, but I don't know if you have notifications on new comments on old issues.

Do you have any update of what has happened and when it will work again?
Best regards,
Anders

andersbjn, 2021-09-17:
Now server is responding with timezone definition on port 2342 so it seems ok!
Thanks!

logowe, 2021-09-17:
**** Still a TIMEOUT now:

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START

ezTime debug level set to INFO
Waiting for WiFi ... Querying pool.ntp.org ... ERROR: No network
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.220, MASK: 255.255.255.0, GW: 192.168.1.1
Querying pool.ntp.org ... success (round trip 52 ms)
Received time: Friday, 17-Sep-21 10:53:20.624 UTC
Time is in sync

UTC:             Friday, 17-Sep-2021 10:53:20 UTC
Timezone lookup for: Switzerland/Zurich ... ERROR: Timeout
Switzerland:     Friday, 17-Sep-2021 10:53:22 UTC

*** When I change Remote port to 4243, the server does respond but timezone is NOT set.

logowe, 2021-09-17:
It also works for my apps again, thanks!