ropg/ezTime #126: nrf.h: No such file or directory - on create.arduino.cc

jfranken01, 2021-09-20:
Hi Rop

I copy and paste your example code (a complete sketch to show how simple it is) into the arduino online editor and hit compile for a Arduino Nano 33 IoT board.

It comes back with a compile time error for "nrf.h: No such file or directory".

Can you help me start with a 'working sketch' to adapt to my existing other initiatives?

Many thanks!
Jan Franken - Canada

jfranken01, 2021-09-20:
when I use the desktop editor and add the ezTime library via the library manager, I also get a compile error, this time for EEPROM.h: No such file or directory

jfranken01, 2021-09-20:
When I install the Arduino_NVM library to support this, it then compile time error similar to the cloud edition with a nrf.h: No such file or directory

mrh335, 2022-05-12:
I am also trying to use a Nano IoT 33 with eztime, but unable to compile. I can't see the detailed error.