ropg/ezTime #127: Can't find Preference.h ESP32 PlatformIO

mars000, 2021-10-03:
hi
I am trying to compile on PlatformIO on ESP32
I have commented in `#define EZTIME_CACHE_NV`S as per instructions.
PlatfromIO still resulting in compile error....can't figure out why :-( I have confirmed `Preferences.h `is in esp32 library
Pls help.
A noob quesiton - where do I find codes for countries/cities ?

[image]

ropg, 2021-10-03:
I haven't done much testing with PlatformIO, just with the Arduino IDE, so can't help much there.

As for timezone names: [here](https://www.zeitverschiebung.net/en/all-time-zones.html) is a list.

MacSass, 2022-02-13:
I have the same issue with ESP32 and PlatformIO - anybody been able to fix this "missing Preferences.h" error?

Gixy31, 2022-07-31:
VSCode (the PlatformIO IDE) doesn't accept conditional code before precompiled headers (such as Preferences.h). You need to modify ezTime.cpp and include all the required files without #ifdef. That means that the modified ezTime.cpp file is no more generic but specific to your platform. Another solution is to modify the VSCode configuration not to use precompiled headers, but this will force complete recompilation each time you make a modification, I didn't try that.