ropg/ezTime #149: hour() is off when passing in a timestamp, or I'm using it wrong?

alager, 2022-03-20:
`Received time: Sunday, 20-Mar-22 14:50:07.359 UTC`
`Time is in sync`
`our Timezone: America/Chicago`
`Timezone lookup for: America/Chicago ... (round-trip 392 ms)  success.`
 ` Olson: America/Chicago`
`  Posix: CST6CDT,M3.2.0,M11.1.0`
`Central Time:     Sunday, 20-Mar-2022 09:50:07 CDT`

Serial << "now: " << myTZ.now() << mendl;
`now: 1647769859`

Serial << "storedTime: " << storedTime << mendl;
`storedTime: 1647769900`

Serial << "timezone: " << myTZ.getTimezoneName( storedTime, LOCAL_TIME ) << mendl;
`timezone: CDT`

uint8_t hour = myTZ.hourFormat12();
Serial << "hour: " << hour << mendl;
`hour: 9`  **<-- the expected value**

uint8_t hourStored = myTZ.hourFormat12( storedTime, LOCAL_TIME );
Serial << "hourStored: " << hourStored << mendl;
`hourStored: 2` **<-- wrong value.  I think it should be either 9, or 4 depending on UTC or local timezone.**

uint8_t minute = myTZ.minute();
Serial << "minute: " << minute << mendl;
`minute: 50`

uint8_t minuteStored = myTZ.minute( storedTime, LOCAL_TIME );
Serial << "minuteStored: " << minuteStored << mendl;
`minuteStored: 51`

k5map, 2022-07-23:
@alager 

I'm experiencing the same problem as you with both the hour() and hourFormat12() functions.  What is even more confusing is if I output hour using the dateTime() function, the hour is correct.  I'm in the CDT timezone and my hour() value is +5 higher than it should be (hour() returns 9 when my local hour is 4).  My timezone and DST are set correctly.

nilton61, 2022-11-25:
I had the same problem with the day(t) function when passing a timestamp in thew t parameter. I started a new issue on this