ropg/ezTime #152: Asia/Jerusalem returns empty posix string

itaysp, 2022-07-04:
```
Timezone lookup for: Asia/Jerusalem ... (round-trip 237 ms)  success.
  Olson: Asia/Jerusalem
  Posix:
```

The same happens if looking for `IL`:
```
Timezone lookup for: IL ... (round-trip 257 ms)  success.
  Olson: Asia/Jerusalem
  Posix:
```

A-R-S-D, 2022-08-14:
Hi, buddy!
Same problem, Any progress?

dolevhadad, 2024-11-03:
A year after I hope it's still relevant 
the only way that I can solve it is to add the summer and winter time manually by setting the Posiix value.
it still needs some finetuning for the changes in our calendar but it can solve the case 

>     SerialPrintln("");
>     timezone.setLocation(F("Asia/Jerusalem"));
>     timezone.setPosix(F("IST-2IDT,M3.5.5/2,M10.5.0/2"));
>     SerialPrintln("Current time Asia/Jerusalem : ");
>     SerialPrintln(timezone.dateTime());

1.  IST-2:
IST is Israel Standard Time, 2 hours ahead of UTC, hence -2.
2. IDT:
IDT is Israel Daylight Time, which is used during the daylight saving period.

3. M3.5.5/2,M10.5.0/2:
M3.5.5/2: Daylight saving time starts in March (3rd month), on the last (5th) Friday (5) at 2:00 AM.
M10.5.0/2: Daylight saving time ends in October (10th month), on the last (5th) Sunday (0) at 2:00 AM.

Output 
> Timezone lookup for: Asia/Jerusalem ... (round-trip 142 ms)  success.
>   Olson: Asia/Jerusalem
>   Posix: 
> Current time Asia/Jerusalem : 
> Sunday, 03-Nov-2024 23:15:52 IST

I also set the NTP server to IL server - set those lines before the setLocation - not necessary 

```
    setServer("il.pool.ntp.org");
    waitForSync();
    setDebug(DEBUG);  // if you want to see the communication to the server and the logs 
```