ropg/ezTime #163: Using ezTime with nodemcu-32s w/64x32 HUb75 LED Matrix

heffnerm2, 2022-12-30:
I've got it able to display time dma_display->print(myTZ.dateTime("m-j-y")); , however i cannot get it to update the string in real time. I tried   
if (secondChanged()) {
dma_display->print(myTZ.dateTime("m-j-y"));
}
 and it overwrites the initial display ever second without erasing the original.

How can i get it to "replace' the original digits with the new ones?

Thanks

Pius171, 2022-12-30:
I will need to see more of your code, in order to help. 
Have you tried printing to the serial monitor, to see if the time changes?

heffnerm2, 2022-12-30:
Serial monitor doesn't change. Here's whole code:
#include <SPI.h>

#include <Wire.h>
#include <Adafruit_SHT31.h>
#include <WiFi.h>
#include <WiFiAP.h>
#include <WiFiClient.h>
#include <WiFiGeneric.h>
#include <WiFiMulti.h>
#include <WiFiScan.h>
#include <WiFiServer.h>
#include <WiFiSTA.h>
#include <WiFiType.h>
#include <WiFiUdp.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <Adafruit_GFX.h>
#include <Adafruit_GrayOLED.h>
#include <Adafruit_SPITFT.h>
#include <Adafruit_SPITFT_Macros.h>
#include <gfxfont.h>
#include <FastLED.h>
#include <ezTime.h>
//#include <TimeLib.h>
//#include "sntp.h"
//#include "time.h"

//Configuration
const int panelResX = 64;      // Number of pixels wide of each INDIVIDUAL panel module.
const int panelResY = 32;     // Number of pixels tall of each INDIVIDUAL panel module.
const int panel_chain = 1;      // Total number of panels chained one to another

//bool secondChanged();

#define myTZ_POSIX  TZ= EST -5 EDT ,M3 .5 .0 /2 ,m10 .5 .0 /3

//bool enableHeater = false;
//uint8_t loopCnt = 0;

//Adafruit_SHT31 sht31 = Adafruit_SHT31();

char ssid[] = "PZRG4";
char pass[] = "erica610";

//float Ct = sht31.readTemperature();
//float Ih = sht31.readHumidity();
//float It = (Ct*1.8)+32;

MatrixPanel_I2S_DMA *dma_display = nullptr;

Timezone myTZ; 

uint16_t myBLACK = dma_display->color565(0, 0, 0);
uint16_t myWHITE = dma_display->color565(255, 255, 255);
uint16_t myRED = dma_display->color565(255, 0, 0);
uint16_t myGREEN = dma_display->color565(0, 255, 0);
uint16_t myBLUE = dma_display->color565(0, 0, 255);

const int defaultBrightness = (40*255)/100;      // full (100%) brightness

  int16_t xOne, yOne;
  uint16_t w, h;

void displaySetup() {
  
  HUB75_I2S_CFG mxconfig(
    panelResX,   // module width
    panelResY,   // module height
    panel_chain    // Chain length
  );

  // If you are using a 64x64 matrix you need to pass a value for the E pin
  // The trinity connects GPIO 18 to E.
  // This can be commented out for any smaller displays (but should work fine with it)
  //mxconfig.gpio.e = 18;

  // May or may not be needed depending on your matrix
  // Example of what needing it looks like:
  // https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/134#issuecomment-866367216
  mxconfig.clkphase = false;

  // Some matrix panels use different ICs for driving them and some of them have strange quirks.
  // If the display is not working right, try this.
  //mxconfig.driver = HUB75_I2S_CFG::FM6126A;

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
}

void setup() {

  Serial.begin(115200);

  WiFi.begin("PZRG4", "erica610");

/*  
 *   Serial.println("SHT31 test");
  if (! sht31.begin(0x44)) {   // Set to 0x45 for alternate i2c addr
    Serial.println("Couldn't find SHT31");
    while (1) delay(1);
  }
  */

  waitForSync();

  while (WiFi.status() != WL_CONNECTED) {
  Serial.print(".");
  delay(1000);
  }

  Serial.print("Attempting to connect to Network named: ");
  Serial.println(ssid);                   // print the network name (SSID);

  Serial.println("");
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

 /*   // Set NTP polling interval to 60 seconds. Way too often, but good for demonstration purposes.
  setInterval(60);

  // Make ezTime show us what it is doing
  setDebug(INFO);*/
 
  myTZ.setLocation(F("America/New_York"));
  Serial.println (F("New York: "));
  Serial.println (myTZ.dateTime ("j-m-y h:i:s A T"));

  displaySetup();
  
  dma_display->clearScreen();
  dma_display->fillScreen(myBLACK);

  dma_display->setTextSize(1);     // size  == 8 pixels high
  dma_display->setTextWrap(false); // N.B!! Don't wrap at end of line

  // Pointers to this variable will be passed into getTextBounds,
  // they will be updated from inside the method 

  dma_display->getTextBounds("1234567890", 0, 0, &xOne, &yOne, &w, &h);
  int xPosition = dma_display->width() / 2 - w / 2 + 1;
  dma_display->setCursor(xPosition, 0);
  dma_display->setTextColor(myBLUE);
  dma_display->print("33f    45%");

  dma_display->getTextBounds("72F    50%", 0, 0, &xOne, &yOne, &w, &h);
  xPosition = dma_display->width() / 2 - w / 2 + 1;
  dma_display->setCursor(xPosition, 8);
  dma_display->setTextColor(myGREEN);
  dma_display->print("73f    45%");

/*  dma_display->getTextBounds("11:15:11am", 0, 0, &xOne, &yOne, &w, &h);
  xPosition = dma_display->width() / 2 - w / 2 + 1;
  dma_display->setCursor(xPosition, 16);
  dma_display->setTextColor(myWHITE);
  dma_display->print(myTZ.dateTime ("g:i:sa"));
  */
  dma_display->getTextBounds("02-21-22", 0, 0, &xOne, &yOne, &w, &h);
  xPosition = dma_display->width() / 2 - w / 2 + 1;
  dma_display->setCursor(xPosition, 24);
  dma_display->setTextColor(myRED);
  dma_display->print(myTZ.dateTime ("m-j-y"));

  //dma_display->fillScreen(myBLACK);

//delay(1000);

}

void loop() {

void events();{

  if (secondChanged()) {
    
  dma_display->getTextBounds("11:15:11am", 0, 0, &xOne, &yOne, &w, &h);
int xPosition = dma_display->width() / 2 - w / 2 + 1;
  dma_display->setCursor(xPosition, 16);
  dma_display->setTextColor(myWHITE);
  dma_display->print(myTZ.dateTime ("g:i:sa"));
    //dma_display->print(myTZ.dateTime("m-j-y"));
} 
}

heffnerm2, 2023-01-02:
[Centered Text 1_3_23.txt](https://github.com/ropg/ezTime/files/10333440/Centered.Text.1_3_23.txt)

Serial monitor doesn't change.

Pius171, 2023-01-02:
@heffnerm2 , your _void event_ function is inside the void loop, you can't create a _function_ inside another _function_

Also, there are some errors in the void loop. Did this code compile?

heffnerm2, 2023-01-03:
I cleaned it up, yes it does compile and run. I get the displays i expected Temp and Humidity inside and outside and time/date, but it does not update time/date

[Centered Text 1_3_23.txt] (https://github.com/ropg/ezTime/files/10339308/Centered.Text.1_3_23.txt)