ropg/ezTime #175: Use ip-api.com to fetch timezone offsets more accurately?

scottchiefbaker, 2023-12-18:
I've had really good luck using ip-api.com to fetch timezone offsets. They offer simple newline support:

```
# Get the TZ offset for the requesting IP
curl http://ip-api.com/line/?fields=offset
```

Or full featured JSON if needed as well:
```
curl http://ip-api.com/json/
```

ropg, 2023-12-18:
I get "Access to ip-api.com was denied / You don't have authorization to view this page." when I click that.

scottchiefbaker, 2023-12-18:
I wonder if it's GeoLocked? Their documentation doesn't say anything about that though.

It's really great, simple API. Been around since 2012, so I don't think it's going anywhere.

ropg, 2023-12-18:
Hah, works now, although I did get the error once more. Odd. But yes, I agree this is a better way to go than having our own server for this.