ropg/ezTime #180: Not maintained anymore?

mrWheel, 2024-09-02:
Well, it seems there is not much activity on this repo anymore.

Thats a pitty!

It made it to platformio but tries to #include <EEPROM.h> which is not available in platformio (is part of the core).

I would like to have a working version for that environment.

ropg, 2024-09-02:
Yup. There's actually quite a few corners where this needs work, including, but not limited to:

* Source of timezone info needs to be something I do not maintain because that turned out to be a pain.
* The unix epoch needs to be timezone-less, which is a rather fundamental rework.
* Support for various platforms needs a rethink because the higher end platforms have their own date and time functionality that isn't crummy anymore like it often was when I started this.

In other words: if this has a future, it is high time for a version 2.0. But the number of people coding significant things on truly limited platforms (like the original Arduinos) further dwindles...

mrWheel, 2024-09-02:
@ropg ,

Wauw! fast response ...!

Thanks. And indeed. There is a lot going on in the world of MCU's.

It's just that I have a [rather large project](https://willem.aandewiel.nl/index.php/2022/11/15/crisis-what-crisis/) that started of for the Arduino IDE and now I try to convert it to PlatformIO IDE.

Found a workaround though: Just copied ezTime folder to the 'lib' folder and removed the '#include "EEPROM.h"' statement. Still have tons of error's (not related to ezTime).

Hope you find the courage and time!

Regards

closed 2024-09-02 by mrWheel