ropg/ezTime #183: setLocation

staffanwillen, 2024-11-09:
I get the wrong time zone!!! fyi lcdPrint below prints on my lcd. "Time Server Sync" works :) But after the setLocation error "Time zone error" which I always get , I seem to end up at GMT or something - anyway one hour ahead of where I am - in Stockholm/Sweden. I have tried "se", GMT-1", "Sweden/Stockholm" but to no avail. Time here is 02:28 but the HawkTZ object below reports 01:28
Please advise!!!
```
  Timezone HawkTZ;

 // Time Server init
  int c = 0;
  lcdPrint(1,1,"Time Server Sync");
  while (!waitForSync(1)) {
    c++;
    lcdPrint(2,1,String(c));
    if (c==10) {
      NVIC_SystemReset();
    }
   }
  lcdClear();
  if (!HawkTZ.setLocation("se")) {
     lcdPrint(1,1,"Time zone error");
     delay(4000);
  }
  lcdPrint(4,1,HawkTZ.dateTime("j/n H:i"));
```

zhoyu458, 2024-11-09:
Yesterday, I experienced a similar issue: no matter what location I set in the code, I always get the same time from Eztime. I tried again this morning, and everything is working fine now.

staffanwillen, 2024-11-09:
Thank you zhoyu458 for commenting. Same here. Working fine again. Guess ezTime hooks up to a server for the time zone thing, that occasionally is offline(?)

ApRhys900, 2024-11-11:
I have a similar issue. It occasionally displays the incorrect time, and upon investigation it appears to be changing to the New Zealand timezone instead of US central time that I want. In 'ezTime.h' I found the lines pertaining to the authors timezone server (timezoned.rop.nl) and decided to try some unskilled hacks.
The default timeout for this service is set at 2000 ms. I changed it to 20ms, which guarantees a timeout for me, and it started displaying the NZ time only, and never my local time. This technique duplicates the issue, I believe. So my next hack was to double the timeout value to 4000ms. It has worked correctly since, but I am still keeping an eye on it before I commit to using ezTime and not a different library. I am not skilled enough to do a more thorough analysis, and I have no idea why a server in the Netherlands would glitch to NZ time.

staffanwillen, 2024-11-12:
...and it happened again this morning. setLocation("se") returns an error and I get GMT (i think) instead.
Not 100% pleased that ezTime has become unreliable. I have a great deal of sketches based on that :/